foxglove/studio #2323: Show Studio version in Studio

jessicaaustin, 2021-12-08:
Maybe I just missed this, but it would be nice to list the version of Studio somewhere in the app itself. 

This will be helpful if a co-worker reports a problem and I want to know what version they are running for the desktop app.

FWIW, the first place I looked was the Help side pane.

amacneil, 2021-12-08:
It's available under Help > Version on Windows/Linux (hit Alt to make the menus appear), or `Foxglove Studio` > `About` on macOS.

I agree that this is hard to find (especially on Windows/Linux). We have talked about always showing the menu bar, and we could also add the version somewhere (at the bottom?) of the Help panel.

jessicaaustin, 2021-12-09:
Ah I didn't realize there was a menu! I like the idea of showing the version at the bottom of the Help panel. Then it shows up in the same place no matter desktop/web or what OS.

amacneil, 2021-12-09:
@esthersweon thoughts on adding to bottom of Help panel?

@jessicaaustin our website doesn't technically have a version (other than the git sha) since it's continuously deployed. Not sure how we want to handle that. Maybe we could display the git sha, it's better than nothing.

closed 2021-12-10 by jtbandes