foxglove/studio #2334: WebSocket Connection has the wrong default value in

KerstinKeller, 2021-12-09:
**Description**
When trying to connect to the websocket server, the default connection in Studio is a link to a bag file
```
https://storage.googleapis.com/foxglove-public-assets/demo.bag
````
instead of 
```
"ws://localhost:8765"
```

- Version: latest dev build, since today

**Steps To Reproduce**
[image]
Try to open the Websocket connection.

jtbandes, 2021-12-10:
I'm having trouble reproducing this issue. Can you try again on the latest version and confirm?

Also FYI, you should be able to use the new url format:

https://studio.foxglove.dev/?ds=foxglove-websocket&ds.url=ws://localhost:8765/

KerstinKeller, 2021-12-10:
It's a bit strange, it seems to happen when I refresh the brower, after a connection was already established...

jtbandes, 2021-12-10:
I still don't see the behavior when I refresh 😕 maybe you can record a video?

KerstinKeller, 2021-12-13:
https://user-images.githubusercontent.com/13848742/145814354-e2f8f869-63ee-43d1-bb10-e8e1a4b6428a.mov

I am hoping this helps.

PR #2369 (2021-12-15): Encode url state for foxglove websocket connections.
**User-Facing Changes**
This fixes url state encoding for foxglove websocket connections.

**Description**
The URL state wasn't being set at all in the `FoxgloveWebSocketPlayer`. This fixes that.

Fixes #2334
closed 2021-12-15 by foxymiles