foxglove/studio #2343: Newly subscribed panels never receive old messages

jtbandes, 2021-12-09:
**Description**
When Studio subscribes to a latched message, it immediately receives the message and displays it in any subscribed panels. However, if panels are added _later_ that subscribe to the same topic, they will not receive the old message even if it is still visible in other panels.

**Steps To Reproduce**
- Run `ros2 launch rosbridge_server rosbridge_websocket_launch.xml` (must use a version incorporating fix from https://github.com/RobotWebTools/rosbridge_suite/pull/689)
- Run `ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 a b`
- Connect Studio via Rosbridge (ROS 2)
- Create a new layout in Studio.
- Add a 3D panel. Under the topic picker, the "TF" section now contains `a` and `b`.
- Remove all panels from the layout.
- Add a Raw Messages panel and subscribe to `/tf_static`. A message appears in the panel.
- Add a 3D panel. Under the topic picker, the "TF" section is empty.

**Expected Behavior**
The static TFs should be available to the 3D panel regardless of which order panels are added.

**Actual Behavior**
Static TFs are only visible if the 3D panel was the first panel to subscribe to `/tf_static`.

jtbandes, 2021-12-09:
Possible fix: Player or Message Pipeline could keep the most recent message on each topic in memory

defunctzombie, 2022-01-11:
I tried this with a local bag file and found that I could open an image panel then a raw message panel for the same topic. I'm thinking the current app behavior for message pipeline sends the subscription for new panels down to the player and if the player had the messages stored then it sends the messages to the new subscriber.

I bring that up to think about how any caching in message pipeline could interfere with subscribing to the player and getting messages from the player. Its interesting to think about handling this generically at the message pipeline layer tho I wonder if we should leave it up to the player.

jhurliman, 2022-01-18:
@defunctzombie thoughts on how to move forward with this? Caching at the message pipeline layer or per player is fine IMO (although the latter will require getting it right in many more places and eventually user-written data source extensions as well).

defunctzombie, 2022-01-28:
I've been trying to come up with a strategy and the only one I've been able to come up with is that its a _per player_ issue.

I started looking at "ok maybe I can make message pipeline hang on to the last message on any topic and when you subscribe you get that" but ran into the above note about interfering with what the player does. For example the random access player has the following logic:

https://github.com/foxglove/studio/blob/04ed2c9144a2f7e83d8c2cb96838b091bb46ff51/packages/studio-base/src/players/RandomAccessPlayer.ts#L598-L618

And this logic is run when the player "seeks". Which happens when a "request backfill" happens. Which happens from the message pipeline when subscriptions change. All of that is messy but describes what happens today. So if we do this at the message pipeline layer then there's a race with the player emitting the new messages and message pipeline emitting messages. Really message pipeline is not there to inject messages - only interface to players.

commit 0717b692 (2022-02-04): Inject the last message on a topic for new subscribers

Fixes: #2343
commit ac57c954 (2022-02-04): Inject the last message on a topic for new subscribers

Fixes: #2343
PR #2778 (2022-02-09): Inject the last message on a topic on new subscriptions

**User-Facing Changes**
When adding a panel which needs to subscribe to an infrequently publishing topic (i.e. /tf_static), the new panel will receive the message.

Previously only the first panel to subscribe would receive the message.

**Description**
This change updates the MessagePipelineProvider to add logic for injecting the last message on a topic to new subscribers of the topic. The provider tracks when new panels subscribe, via setSubscriptions calls, and makes a note on any new topics requested by the panel. When a new player s [... cut]
commit 319b2734 (2022-02-09): Inject the last message on a topic on new subscriptions (#2778)

**User-Facing Changes**
When adding a panel which needs to subscribe to an infrequently publishing topic (i.e. /tf_static), the new panel will receive the message.

Previously only the first panel to subscribe would receive the message.

**Description**
This change updates the MessagePipelineProvider to add logic for injecting the last message on a topic to new subscribers of the topic. The provider tracks when new panels subscribe, via setSubscriptions calls, and makes a note on any new topics requested by the panel. When a new  [... cut]
closed 2022-02-09 by defunctzombie