foxglove/studio #2365: Live monitoring of Behavior Tree for many mobile robots

ljaniec, 2021-12-14:
Foxglove should support live monitoring of Behavior Trees execution live during navigation of many robots, as a custom panel or extension (similar to [Groot and Nav2 stack](https://navigation.ros.org/tutorials/docs/using_groot.html), but with adequate continuous support for debugging and new features).

Shortly: Since ROS Humble there will be no option to see BT during execution of the BT Navigator from Nav2 stack (ZeroMQ bugs for multirobot systems, it was described in some issue and in  "Migration from ROS Galactic to Humble" notes)

- https://github.com/ros-planning/navigation2/issues/2386
- https://navigation.ros.org/migration/Galactic.html?highlight=multi#bt-navigator-groot-multiple-navigators
- https://navigation.ros.org/migration/Galactic.html?highlight=multi#dropping-support-for-live-groot-monitoring-of-nav2

I see that an option to view in real-time execution of 1+ BT will be an awesome new feature in Foxglove, because since next year and Humble there will be no clear candidate for this.

Best,
Łukasz Janiec

jtbandes, 2021-12-14:
Hi, thanks for sharing your feature request!

We are not very familiar with Behavior Trees and Groot. Do you know how popular these tools/libaries are or how many people use them?

It looks like they are currently unmaintained: https://github.com/BehaviorTree/BehaviorTree.CPP / https://github.com/BehaviorTree/Groot

> No company, institution or public/private funding is currently supporting the development of BehaviorTree.CPP and Groot. As a consequence, my time to support BehaviorTree.CPP is very limited and I decided that I won't spend any time at all supporting Groot. Pull Requests are welcome and will be reviewed, even if with some delay.

Which features from Groot are most important to you and would be required for a minimum viable solution?

ljaniec, 2021-12-15:
Behavior Trees are important tool for robotics, especially ROS 2 and its Navigation2 (Nav2 stack). 

Examples of active, maintained packages with use of BT:
- https://github.com/ros-planning/navigation2/tree/main/nav2_behavior_tree
- https://github.com/ros-planning/navigation2/tree/main/nav2_bt_navigator
- https://github.com/Adlink-ROS/BT_ros2

Short intro to BT in Nav2:
- https://navigation.ros.org/concepts/index.html#behavior-trees
- https://navigation.ros.org/tutorials/docs/using_groot.html

There are `nav2_bt_navigator` in the development too. Groot repository isn't actively maintained, that's why I would like to have other tool for visualization and monitoring of Behavior Trees in ROS2 projects (on simulated and real robots). Foxglove fits this niche perfectly.

Minimum viable solution/features would be:
- display a Behavior Tree like that (picture from Groot with default Nav2 BT):
[image]
- monitor a Behavior Tree of chosen robot (with option of choice which one):
[image]
(picture from https://robohub.org/introduction-to-behavior-trees/)

jtbandes, 2021-12-15:
Can you provide or link to some sample bags which contain this data?

ljaniec, 2021-12-16:
Yes, of course:
- https://navigation.ros.org/behavior_trees/index.html and https://navigation.ros.org/behavior_trees/overview/nav2_specific_nodes.html
- https://navigation.ros.org/configuration/packages/configuring-bt-xml.html#configuring-behavior-tree-xml (at the bottom of page) and on GitHub https://github.com/ros-planning/navigation2/tree/main/nav2_behavior_tree
- https://navigation.ros.org/behavior_trees/overview/detailed_behavior_tree_walkthrough.html (and `Navigate To Pose With Replanning and Recovery` BT - picture and XML below)

jtbandes, 2021-12-16:
Is the BT itself stored in the bag file? And which other message types are used to indicate the current BT state?

ljaniec, 2021-12-17:
AFAIK BT isn't stored in a rosbag (it was mentioned [there](https://answers.ros.org/question/344715/groot-how-finds-models-how-connects-to-running-bt/)). It seems that BT state could be indicated through [currently executed plugin](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/pull/27) or nodes return values (SUCCESS/RUNNING/FAILURE). Honestly, I don't have any deeper knowledge about it :( I linked this feature request in Nav2 Slack, maybe other more knowledgeable people could help more.

adamconkey, 2022-11-22:
I am very interested in this feature, and I think it would be a great addition to Foxglove Studio in general since BehaviorTree.CPP has become the defacto behavior tree tool for robotics, and Groot is its frontend graphical tool for editing/monitoring the trees. Things seem to be a bit in flux, [Groot](https://github.com/BehaviorTree/Groot) development has been suspended and only works with BehaviorTree.CPP v3.8 and will soon be replaced by [Groot2](https://github.com/BehaviorTree/Groot2) that will support the new BehaviorTree.CPP v4. I think the new Groot/BehaviorTree.CPP versions might not ensure back-compatibility, so it'd be wise to target development for the new ones once Groot2 is released.