foxglove/studio #2378: Clicking in Diagnostic Summary of demo layout does not reuse the existing Diagnostic Details panel

jhurliman, 2021-12-14:
[NuDemo-001.json.zip](https://github.com/foxglove/studio/files/7715253/NuDemo-001.json.zip)

Import the attached layout, open any bag with a `/diagnostics` topic, and switch to the fourth tab ("Diagnostics"). Click on a diagnostic in the Diagnostic Summary panel. Instead of reusing the existing Diagnostic Detail panel it will open a new panel.

jtbandes, 2021-12-15:
Behaving as designed due to an inherent limitation of the mosaic layout.

A layout is a binary tree of panels. The "update existing panel" logic only occurs if the panels are siblings. In your example layout, they have an...avuncular relationship? Anyway, here is a modified version of the layout where it works:

[NuDemo-001.json.zip](https://github.com/foxglove/studio/files/7716000/NuDemo-001.json.zip)
 
It might make sense to leave this issue open but I think it needs to be re-conceptualized as a new feature and possibly complete rework of the mosaic structure (there are other use cases that might benefit from degree 2+ mosaic trees).

defunctzombie, 2022-06-24:
> I think this could be fixed by traversing the entire tree (starting at the current tab if there is a tab panel parent) and looking for a Diagnostic Details panel to update before creating a new one.

A user might not always want any diagnostic detail panel to update tho. They might have crafted one for a specific diagnostic and want to keep that.

I'm actually not so much a fan of the linked diagnostic panels - somehow I'd almost prefer it to be a mode of the panel itself rather than a separate panel.

jtbandes, 2022-06-25:
Duplicate of https://github.com/foxglove/studio/issues/3675, fixed in https://github.com/foxglove/studio/pull/3687

closed 2022-06-25 by jtbandes