foxglove/studio #2379: Diagnostic Detail panel dropdown hides all other options when one is selected

jhurliman, 2021-12-14:
The Diagnostic Detail panel has a dropdown that allows showing the key/values + OK/ERROR status for all diagnostics or only a single diagnostic. The "all diagnostics" option is just a blank dropdown entry which looks more like a bug than a purposeful choice, and the behavior of the dropdown is different after selecting the blank option vs one of the other options. Selecting a specific option and then opening the dropdown again, it looks like it's using the entry name as filter text so no other options appear.

[image]
[image]

jtbandes, 2021-12-15:
This seems to depend which kind of item you click. If you click an item that represents a whole hardware_id, the problem doesn't occur. If you select a hardware_id+name combo, it does. For example with this data (from demo.bag):

[image]

If I choose the `piksi rtk` item, I don't experience the problem, but I do with the other two more specific piksi options.

amacneil, 2021-12-20:
Need to discuss desired behavior.

defunctzombie, 2022-01-14:
I re-read the original issue description and I do not understand how to reproduce. @jhurliman could you indicate what steps you took to get into the state where you see blank entries? I agree what I see in the screenshot with blank dropdown entries looks undesirable. What I don't yet understand is what the desired behavior is or the steps to reproduce.

jhurliman, 2022-01-14:
Here's a video demonstrating the problem. You can reproduce this by playing around with the dropdown in the fourth tab of the nuScenes layout.

https://user-images.githubusercontent.com/195374/149594924-db6f7497-b531-4f41-af6f-01d3c6245099.mov

Desired behavior:
 - Don't use `""` (empty strings) as a dropdown entry. It looks accidental and doesn't clue the user as to what will happen if you select it
 - Don't hide all of the other dropdown items when one is already selected

commit 5d8e100c (2022-01-19): Improve usability for Diagnostic Detail panel

The autocomplete in the Diagnostic Detail panel would hide available options once the user selected one. To change to a different diagnostic the user has to click the input, realize they can delete their text, delete all the text and type again to find their entry.

This change switches the autocomplete input to the mui Autocomplete component. This component exhibits better UX behavior. When focusing on the input, the user is shown all available entries with their current entry highlighted. This also resolves some glitches with the dropdown.

This [... cut]
commit 123c231c (2022-01-19): Improve usability for Diagnostic Detail panel

The autocomplete in the Diagnostic Detail panel would hide available options once the user selected one. To change to a different diagnostic the user has to click the input, realize they can delete their text, delete all the text and type again to find their entry.

This change switches the autocomplete input to the mui Autocomplete component. This component exhibits better UX behavior. When focusing on the input, the user is shown all available entries with their current entry highlighted. This also resolves some glitches with the dropdown.

This [... cut]
commit 5e5ee0ae (2022-01-19): Improve usability for Diagnostic Detail panel

The autocomplete in the Diagnostic Detail panel would hide available options once the user selected one. To change to a different diagnostic the user has to click the input, realize they can delete their text, delete all the text and type again to find their entry.

This change switches the autocomplete input to the mui Autocomplete component. This component exhibits better UX behavior. When focusing on the input, the user is shown all available entries with their current entry highlighted. This also resolves some glitches with the dropdown.

This [... cut]
PR #2684 (2022-01-20): Improve usability for Diagnostic Detail panel
**User-Facing Changes**
The Autocomplete widget is replaced with a different autocomplete widget. The overall experience is similar to before (you can type to filter the list). The new widget resolves some UX strangeness from the previous widget described in more detail below.

We've also removed the code paths which strip away a leading "/" from the hardware id as well as the code paths which remove the hardware_id string from the _name_ field. Both of these looked like webviz "cruisisms" and don't seem to be something that should be done for all  [... cut]
closed 2022-01-20 by jtbandes