foxglove/studio #2383: Changing link preference to "Ask each time" immediately goes back to launch choice screen

jtbandes, 2021-12-14:
1. Visit https://studio.foxglove.dev/?ds=foxglove-websocket&ds.url=ws://localhost:8765/
2. Select "remember my preference" and click "Web"
3. Open Preferences sidebar and select "Open links in: Ask each time"

**Expected:** no change for the current session; the **next** time I open a link, I will be asked whether to open it in web or desktop.

**Actual:** the app immediately reverts back to the screen where I originally clicked Web for the current session.

https://user-images.githubusercontent.com/14237/146097368-08d013b5-49e2-42d3-b00c-f12ef9c0d898.mov

foxymiles, 2021-12-21:
I'm having trouble reproducing this one. Any other steps I might need to take @jtbandes ?

jtbandes, 2021-12-21:
Nope, same repro steps still work for me...

foxymiles, 2021-12-21:
> Nope, same repro steps still work for me...

Do you see any difference in an incognito vs normal window?

foxymiles, 2021-12-21:
Note: this seems to only happen if studio can't connect to the websocket and start up a player.

PR #2463 (2022-01-03): Fix handling of launch preference change during web session.
**User-Facing Changes**
This fixes handling of launch preference changes in an active session.

**Description**
The solution here is to set a  temporary, `sessionStorage` preference for web if the user changes the launch preference to "ask me" during an active web session so that the existing session isn't disrupted.

Fixes #2383 
Fixes #2511
closed 2022-01-03 by foxymiles