foxglove/studio #2395: Remove claims of supporting package:// URLs on web

amacneil, 2021-12-15:
**Description**
On web + desktop, for URDF support we claim to support `package://` URLs. This feature only works on desktop.

Using a `package://` URL results in an error silently being logged to the developer console:

https://user-images.githubusercontent.com/637671/146259462-60205d79-9b65-40d8-b075-ca1c88643710.mov

In this dialog, on desktop we can list support for both methods (package and http(s). On web, we should explain that only http(s) is supported and that they can get package support by installing the desktop app.

We should also explain in this dialog how `ROS_PACKAGE_PATH` is important for package URLs and maybe even display the file we are going to try to load.

cc @defunctzombie for thoughts on the ux

defunctzombie, 2021-12-15:
I don't recall if we have a good way to detect platform capabilities (i.e. whether we can support package:// paths or not). We might have some hard-coded util.js file which might be a decent(ish) place to start with the detection.

defunctzombie, 2021-12-16:
I've assigned this to myself since John has lots of `tf` and `urdf` related issues to look at already.

PR #2412 (2021-12-20): Omit package:// references from URDF settings editor in web

**User-Facing Changes**
Users are not shown a hint suggesting package:// urls are valid in the web app.

**Description**
Web does not support loading package:// urls so we do not display this as a hint to the user.

Fixes: #2395
commit ffc67412 (2021-12-20): Omit package:// references from URDF settings editor in web (#2412)

**User-Facing Changes**
Users are not shown a hint suggesting package:// urls are valid in the web app.

**Description**
Web does not support loading package:// urls so we do not display this as a hint to the user.

Fixes: #2395
closed 2021-12-20 by defunctzombie