foxglove/studio #2458: Foxglove desktop app doesn't show subscribing topics when connected directly or via rosbridge

s-sliwinski, 2021-12-21:
**Description**
Foxglove desktop app doesn't show subscribing topics when connected directly and doesn't allow to publish to them

- Version: v0.24.0
- OS: Ubuntu 20.04.3 LTS (Focal Fossa)

**Steps To Reproduce**

I'm using RPI4 as my ROS machine. There i run roscore and subscriber node (/listener) from this tutorial: http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29.
But the /chatter topic doesn't show up in foxglove. I must add that it displays in terminal on PC connected to raspberry so the connections seams to be ok.

## rostopic list:
[image]

## roswtf:
[image]

## foxglove direct connection:
[image]

## foxglove rosbridge connection:
[image]

## rosbridge connection roswtf:
[image]

**Expected Behavior**
/chatter topic displays in "Raw Message" and "Publish" panels. 

**Actual Behavior**
/chatter topic doesn't show up in foxglove studio

When i enter /chatter directly in the "Publish" panel and publish the message shows in foxglove studio but doesn't go to roscore:
[image]

amacneil, 2021-12-24:
Often this is a problem where the ROS host (robot) doesn't know the correct IP address or hostname of your computer, which it needs to publish messages to the client (in this case the "client" is Foxglove Studio).

In v0.24 you can solve this by setting `ROS_HOSTNAME` in the settings sidebar (cog icon in bottom right):

[image]

In v0.25 we have made this more prominent, you will see it in the new Open dialog whenever you initiate a ROS 1 connection:

[image]

Let me know if that helps! If not, we can try additional debugging steps.

defunctzombie, 2022-01-06:
I also echo what @amacneil said above. Please give the new connection dialog a try and see if that helps clarify how to connect to a live ROS1 system.

If you believe you setup the ROS_HOSTNAME and ROS_MASTER_URI correctly in Studio and your robot feel free to re-open this issue and we can take a look again.

closed 2022-01-06 by defunctzombie