foxglove/studio #2459: Incorrect visualization of mono16 or 16UC1 images

gilduz, 2021-12-21:
**Description**
16 bit images with encoding equal to mono16 (i.e. grayscale) or 16UC1 (i.e. depth images) are overexposed. Probably just the LSB part of the 16 bit pixel value is visualized.

- Version: 0.22
- OS: Ubuntu 18.04

**Steps To Reproduce**

Publish a 16 bit image with the following header:
`header: 
  seq: 1
  stamp: 
    secs: 1640088816
    nsecs: 948826649
  frame_id: "grayscale"
height: 480
width: 640
encoding: "mono16"
is_bigendian: 0
step: 1280
`
or 
`header: 
  seq: 1
  stamp: 
    secs: 1640088816
    nsecs: 948826649
  frame_id: "depth"
height: 480
width: 640
encoding: "16UC1"
is_bigendian: 0
step: 1280
`

**Expected Behavior**
Visualization of the HSB pixel byte.

**Actual Behavior**
Visualization of the LSB pixel byte.

jtbandes, 2021-12-21:
You might be experiencing issues with the min/max value settings. The default max value is 10000, which we inherited from the ROS `image_view` tool: https://github.com/foxglove/studio/blob/932584ce379f4231753bb116f76541bd60708a01/packages/studio-base/src/panels/ImageView/decodings.ts#L141

You can change the settings here:

[image]

[image]

gilduz, 2021-12-24:
Great! I didn't know about the possibility to customize the range. This works.

closed 2021-12-24 by gilduz