foxglove/studio #2474: Topic error for missing transform(s) is misleading when the transform from render frame to fixed frame is missing

jhurliman, 2021-12-22:
**Description**

Now that we round trip transforms through a fixed frame, the code that generates the error message about missing transforms needs to become more intelligent. This is what I see right now:

[image]

"missing to base_link" is correct. "base_link to base_link" and "sensor to base_link" being missing is confusing, it really means "base_link to map" is missing.

defunctzombie, 2022-01-06:
Related issue: https://github.com/foxglove/studio/issues/2575

defunctzombie, 2022-01-06:
Removing the `bug` label from this and keeping `ux` label. This issue and #2575 would both help end users understand what our new tf interpolations are doing under the hood and why they are seeing errors.

closed 2022-06-23 by jhurliman