foxglove/studio #2476: Incorrect / mutated data is shown when clicking on markers in the 3D panel

jhurliman, 2021-12-22:
Clicking on a marker in the 3D panel brings up an overlay with a tab named "Selected object" that shows a view similar to the Raw Message panel for the selected marker. However, instead of showing the original raw marker message, we display the mutated "renderable" marker that is sent to Worldview. This is a confusing thing to see, since the header.frame_id might be `map` but if we're rendering in `base_link` the pose field will be relative to `base_link` and not `map`, and dynamically updating instead of showing the original message contents.

It looks like there is some leftover concept of `originalMessage` in various parts of the code, but it's not wired up to anything that I can see.

[image]

defunctzombie, 2022-01-06:
@jhurliman Are fields like "settings" also not meant to be present? What's a good way to identify that the pose (or other fields that should be present) are correct? Their values should match the same marker value from the Raw Message panel?

jhurliman, 2022-01-07:
@defunctzombie as a user, I would expect clicking on a marker to show the message that created this marker. Or maybe a subset of the message that created this marker, like a single Marker in a MarkerArray. Right now, it shows an object that is internal to Studio and not documented anywhere, with fields that don't exist on any message like "settings" and the value of pose mutating every frame (or a pose field injected even when the initial message didn't have a pose field).

PR #2617 (2022-01-11): Remove internal studio fields from 3d panel selected object details popup 
**User-Facing Changes**
Selected object data in the 3d panel shows information about the selected object or marker free of any internal implementation fields (i.e. settings, hitmap, etc).

Previously these Studio implementation specific fields would pollute the selected object display resulting in noise and confusion with the user's dataset.

| Before | After |
| --- | --- |
| [image] |[image]|

**Description**
Fixes: #2476
commit 18ee3913 (2022-01-11): Remove internal studio fields from 3d panel selected object details popup  (#2617)

When a user selects an object in the 3d panel, the object details panel opens with details on the object. This panel would confusingly show internal implementation fields alongside the marker fields.

This change updates the display to only show fields from the original marker message.

Fixes: #2476
closed 2022-01-11 by defunctzombie