foxglove/studio #2481: "Data sources" sidebar, Duration should show nanoseconds when nanosecond-level timestamps are enabled

jhurliman, 2021-12-22:
Studio allows toggling between "human-friendly" time and "whole_seconds.fractional_seconds" formatting. When the latter is selected, the "Duration" field of the Data sources sidebar should show full nanosecond precision instead of rounding to the nearest millisecond.

[image]

jtbandes, 2021-12-22:
I personally don't think it particularly makes sense to tie the time format directly to the seconds' precision. They could be two separate settings or we could just always show nanosecond precision.

The screenshot you showed is also weird because it shows "date > time", but the time is an absolute time rather than being relative to the date...

jtbandes, 2022-01-08:
Removing bug label – this issue doesn't really render the app broken.

PR #2628 (2022-01-11): Render DataSourceInfo Duration using app time format

**User-Facing Changes**
Duration values in the Data Source Sidebar and Data Source Info panel conform to the time format application setting. For _seconds_ the value is displayed as decimal seconds. For _time-of-day_ the value is displayed as h:mm:ss.sssssssss.

[image]
[image]
[image]

**Description**
Rendering the duration on the DataSourceInfo sidebar or panel uses the global app time format. If the time format is seconds, then the duration is displayed as decimal seconds. If the time format is time-of-day, then the duration is displayed [... cut]
commit b9b5061a (2022-01-11): Render DataSourceInfo Duration using app time format (#2628)

Rendering the duration on the DataSourceInfo sidebar or panel uses the global app time format. If the time format is seconds, then the duration is displayed as decimal seconds. If the time format is time-of-day, then the duration is displayed as h:mm:ss.sssssssss.

Fixes: #2481
closed 2022-01-11 by defunctzombie