foxglove/studio #2507: Data source start/end time are truncated without ellipsis

jtbandes, 2021-12-23:
When the data source sidebar is narrow, the data source name has an ellipsis in the middle, but the start/end time are just cut off.

[image]

foxymiles, 2022-01-07:
@jtbandes We have a lot of vertical space to work with here. Should we wrap instead of truncating?

jtbandes, 2022-01-07:
cc @2metres

jtbandes, 2022-01-14:
Looks better, what does it do when you get even narrower?

foxymiles, 2022-01-14:
> Looks better, what does it do when you get even narrower?

[image]

This is as narrow as it will let me go.

PR #2676 (2022-01-19): Wrap datasource timestamps in narrow views.
**User-Facing Changes**
This improves rendering of datasource timestamps in narrow views.

**Description**
The approach here is to just wrap the text.

[image]

Fixes #2507
closed 2022-01-19 by foxymiles