foxglove/studio #2508: Error loading STL files

esthersweon, 2021-12-23:
**Description**
- Loading the following `.stl` files in Foxglove Studio does not correctly display the mesh resource in the 3D panel.
   - https://gist.githubusercontent.com/k-okada/d917b82418dcbcbdb478bb106f7ae0ff/raw/c10fb1f508d2cef58d1fbb601edd562f53624154/box.stl
   - https://raw.githubusercontent.com/turtlebot/turtlebot/melodic/turtlebot_description/meshes/stacks/hexagons/plate_simple.stl
 
   
- Version: 0.24
- OS: macOS

**Steps To Reproduce**

- Example bag: https://drive.google.com/file/d/16RR4ijj7_mMr8l3-wvjMmQtXnk8FF3y2/view?usp=sharing
- Use Node Playground to replace the existing the `/avocado` topic's existing `mesh_resource` field with the above `.stl` files

```
import { Input, Messages } from "ros";

type Output = Messages.visualization_msgs__Marker;
type GlobalVariables = { id: number };

export const inputs = ["/avocado"];
export const output = "/studio_node/another_resource";

// Populate 'Input' with a parameter to properly type your inputs, e.g. 'Input<"/your_input_topic">'
const publisher = (
  message: Input<"/avocado">,
  globalVars: GlobalVariables
): Output => {
  return {
    ...message.message,
    mesh_resource:
      "https://gist.githubusercontent.com/k-okada/d917b82418dcbcbdb478bb106f7ae0ff/raw/c10fb1f508d2cef58d1fbb601edd562f53624154/box.stl",
  };
};

export default publisher;
```

- Toggle on the new topic (/studio_node/another_resource) in the 3D panel's topic picker
[image]

**Expected Behavior**
- You see the new mesh resource in the 3D scene

**Actual Behavior**
- No new mesh resource appears in the 3D scene

jtbandes, 2021-12-23:
Are any errors logged in the console? It would be helpful to include that info in this ticket to help with triage.

esthersweon, 2022-01-03:
@jtbandes - Added.

jhurliman, 2022-01-05:
@esthersweon I'm not planning on learning regl-worldview mesh rendering internals any time soon. I assigned this to @jtbandes who might have insight there.

closed 2022-01-06 by jtbandes