foxglove/studio #2512: Can't select child topics when the parent topic is unselected

amacneil, 2021-12-23:
**Description**

https://user-images.githubusercontent.com/637671/147292182-ffe29bb3-016c-4237-a0d8-6b99c2a86338.mov

foxymiles, 2022-02-15:
What's the desired behavior here? Should it also select the parent topic or do we want to make it possible to select child topics without selecting the parent?

amacneil, 2022-02-15:
IMO: If a parent topic is deselected, then selecting a child topic should automatically select its parent topic, and deselect any sibling topics.

defunctzombie, 2022-09-06:
Added the new3d panel so we can take a look at whether this is still relevant for the 3d (beta) panel.

amacneil, 2022-09-08:
New 3D solves this by removing the "disable group" function in favor of "show all" / "hide all", which is what I think I was advocating for here.

closed 2022-09-08 by amacneil