foxglove/studio #2513: Ignore leading slash in transform names

amacneil, 2021-12-23:
**Description**
Apparently ROS treats `base_link` and `/base_link` as the same transform. However, we treat these differently. Leading to errors like this:

[image]

defunctzombie, 2021-12-23:
I don't know that we need to ignore that. IMO thats ROS ism and other frameworks might not have such behavior.

amacneil, 2021-12-23:
We discussed today, and can discuss again live if you want.

It's definitely a ROSism, but if things work out of the box in RViz (we should verify this first) it would be nice if they also work out of the box in Studio.

One way to implement this we discussed would be to either make it player specific (not perfect for example with data platform which might be ROS or non-ROS), or a better idea was to ignore leading slash when we receive transform data via a ROS TFMessage, but when we receive transform data via a yet-to-be-defined Foxglove Transform message, we disable this legacy behavior.

defunctzombie, 2021-12-23:
I think we should verify what happens in rviz and if doing this in ROS results in warnings or other errors telling the user they've produced miss-matched frames.

I agree having things work out of the box is a good target - but datasets can also be broken so we should be careful about patching over broken data vs helping surface broken data. Part of the tool is to also debug when there are issues with your messages - and this points to a possible issue with messages.

jtbandes, 2021-12-24:
> if doing this in ROS results in warnings or other errors telling the user they've produced miss-matched frames

I wouldn't expect a warning or error. This is just how name resolution works in ROS (at least for node/topic names — maybe different for TF)

defunctzombie, 2022-01-06:
I am removing the bug label. I do not think this is a bug. The app shows that you have different frame names and helps you identify that your robot frames do not match. To me a frame without a leading `/` could indicate that the node was not namespaced properly.

I think its worth thinking about whether this is an enhancement or setting we want to provide or document the behavior we explicitly want to support.

amacneil, 2022-07-06:
@jhurliman is this issue solved by https://github.com/foxglove/studio/pull/3717?