foxglove/studio #2523: Support custom image rotation in ImageView panel

amacneil, 2021-12-24:
### Discussed in https://github.com/foxglove/studio/discussions/2469

<div type='discussions-op-text'>

<sup>Originally posted by **marijapili** December 22, 2021</sup>
Is it possible to implement a feature for image rotation - to adjust the view if the camera is, for example, upside down?</div>

foxymiles, 2022-01-07:
@amacneil Would this be something we'd expose in the settings for the panel? Would we support rotating by any angle or just increments of 90°?

jtbandes, 2022-01-07:
I think 90° rotations and horizontal/vertical flip would be nice. Probably ideal for them to be exposed as buttons in the panel UI, if we have space for it.

foxymiles, 2022-01-07:
@jtbandes Now that we have an expanding toolbar in the upper right to support selectable image markers we could add more controls there.

amacneil, 2022-01-08:
Panel settings might be a better place for it since it's not something you would be regularly changing?

PR #2634 (2022-01-12): Support image rotation and mirroring in the image panel.
**User-Facing Changes**
This implements rotation and both horizontal and vertical mirroring in the image panel.

**Description**
This exposes settings for rotation and mirroring in the image view settings panel. The implementation is straightforward. We just pass the additional parameters to the image worker that does the rendering. The only tricky consideration is that the calculation for the viewbox and zoom fit & fill modes needs to take rotation into account.

This also slightly refactors & tightens up the typing in the image renderi [... cut]
closed 2022-01-12 by foxymiles