foxglove/studio #2543: Per-Pose header is ignored when visualizing `nav_msgs/Path` in the 3D panel

jhurliman, 2021-12-26:
**Description**

In the "Nav Msgs Path" story, both lines should go up and down in a sawtooth if the per-Pose `frame_id` is being respected, and the whole path increase on the Y-axis (to point diagonally away from the camera) if `stamp` is being respected for transform interpolation.

[image]

defunctzombie, 2022-01-27:
@jtbandes @jhurliman could one of you take this.

jtbandes, 2022-02-02:
I wrote a python script to create a bag file with the same data. It seems like Rviz displays the path in the same way. Are we sure the per-pose headers are meant to be respected individually?

https://answers.ros.org/question/367021/which-frame_id-field-in-nav_msgspath-should-be-used/
https://github.com/ros-visualization/rviz/issues/1267#issuecomment-534470381

[image]

defunctzombie, 2022-02-02:
Catting with John in slack about it - and it wasn't clear that they are suppose to be respected. One idea we considered is showing a topic error when we detect the user has done this (different frames per entry) since they might have something in mind that cannot be or is not supported.

jtbandes, 2022-02-04:
What should be our course of action with this ticket? Close as expected behavior?

amacneil, 2022-02-07:
Discussed live. Decided resolution: Our current behavior is "correct" in that it matches rviz, but we should add a topic warning if a `geometry_msgs/PoseStamped` has non-empty and different frame_id from the parent `nav_msgs/Path`

closed 2022-03-30 by jhurliman