foxglove/studio #2545: Loading a link from data platform does not populate the plot with data

defunctzombie, 2021-12-27:
When I open a data platform link in a layout which includes a plot panel I had hoped to be able to see a plot of my data over the entire range I've opened in Studio. Instead I have to start "playing" before the plot panel displays any data (similar to a live connection).

This might be a regression with the data platform data source in Studio.

In adding this feature we should also add subscriber hints. A panel can indicate that a topic should be subscribed to greedily (or whatever hint word we want to use). This would tell the underlying data source to opportunistically fetch the topic beyond the short buffer. This will help us prioritize loading the greedy data before loading the topics that only need the "current" message to display.

jtbandes, 2022-01-03:
> This might be a regression with the data platform data source in Studio.

Nope, never implemented: #1964

jtbandes, 2022-01-31:
Duplicate of #1964

closed 2022-01-31 by jtbandes