foxglove/studio #2551: Cannot advertise or publish from extensions api when using ROS1

ccpowers, 2021-12-29:
When using the extensions API to create an extension with a publisher, the publisher is never advertised, and nothing is published, when using a ROS 1 connection. Switching to using a rosbridge connection allows the publisher to be advertised and publish.

- Version: 0.19.0, Linux desktop app
- OS: Ubuntu 16.04

**Steps To Reproduce**
1. Create an example extension panel following the docs here: https://foxglove.dev/docs/extensions/getting-started
2. Add a button that publishes "Hello world!" with the following modifications to `ExamplePanel.tsx`:
```
import { PanelExtensionContext, RenderState } from "@foxglove/studio";
import { useLayoutEffect, useEffect, useState, useCallback } from "react";
import ReactDOM from "react-dom";

function PublisherExample({ context }: { context: PanelExtensionContext }): JSX.Element {
  const [renderDone, setRenderDone] = useState<(() => void) | undefined>();

  // We use a layout effect to setup render handling for our panel. We also setup some topic subscriptions.
  useLayoutEffect(() => {
    // The render handler is run by the broader studio system during playback when your panel
    // needs to render because the fields it is watching have changed. How you handle rendering depends on your framework.
    // You can only setup one render handler - usually early on in setting up your panel.
    //
    // Without a render handler your panel will never receive updates.
    //
    // The render handler could be invoked as often as 60hz during playback if fields are changing often.
    context.onRender = (renderState: RenderState, done) => {
      // render functions receive a _done_ callback. You MUST call this callback to indicate your panel has finished rendering.
      // Your panel will not receive another render callback until _done_ is called from a prior render. If your panel is not done
      // rendering before the next render call, studio shows a notification to the user that your panel is delayed.
      //

      console.log(`Got render state: ${renderState}`);
      // Set the done callback into a state variable to trigger a re-render.
      setRenderDone(done);
    };

    // After adding a render handler, you must indicate which fields from RenderState will trigger updates.
    // If you do not watch any fields then your panel will never render since the panel context will assume you do not want any updates.

    // tell the panel context that we care about any update to the _topic_ field of RenderState
    context.watch("topics");

    // tell the panel context we want messages for the current frame for topics we've subscribed to
    // This corresponds to the _currentFrame_ field of render state.
    context.watch("currentFrame");

    // subscribe to some topics, you could do this within other effects, based on input fields, etc
    // Once you subscribe to topics, currentFrame will contain message events from those topics (assuming there are messages).
    context.subscribe(["/some/topic"]);

    if (context.advertise) {
      console.log(`About to advertise example topic`);
      try {
        context.advertise("/foxglove/example_publisher", "std_msgs/String");
      } catch (err) {
        console.log(`Caught err while advertising: ${err}`)
      }
    } else {
      console.log('Cannot advertise example publisher');
    }
  }, []);

  // invoke the done callback once the render is complete
  useEffect(() => {
    renderDone?.();
  }, [renderDone]);

  const onButtonClick = useCallback(()=> {
    if (context.publish) {
      console.log(`About to publish from example publisher`);
      try {
        context.publish("/foxglove/example_publisher", {"data": "Hello world!"});
      } catch (err) {
        console.log(`Got error while publishing: ${err}`);
      }
    } else {
      console.log(`Example publisher cannot publish`);
    }
  }, [context]);

  return (
    <>
      <div>Click button to publish</div>
      <button onClick={onButtonClick}>Publish Message</button>
    </>
  );
}

export function initExamplePanel(context: PanelExtensionContext) {
  ReactDOM.render(<PublisherExample context={context} />, context.panelElement);
}
```
3. Load for local development: `npx yarn local-install`
4. Open foxglove studio and load the example panel. Use a local ROS 1 connection (`localhost:11311` or your preferred master port)
5. Echo `/foxglove/example_publisher` 
6. Click the button. Note, no messages published.
7. Switch to using a rosbridge connection. Make sure you have a rosbridge server up and running.
8. Click the button again. "Hello world!" will publish on the example topic.

**Expected Behavior**
Publisher is advertised and able to publish regardless of connection type.

**Actual Behavior**
Publisher does not work with ROS1, only rosbridge.

This may be related to https://github.com/foxglove/studio/issues/2057, in that it also is about publishers working in rosbridge but not ROS1.

defunctzombie, 2021-12-30:
@jhurliman Assigned this one to you since it does sound related to #2057. If we learn otherwise we can re-assign or investigate separately.

foxymiles, 2022-01-04:
@ccpowers I'm working on a better dev experience solution for this but it looks like there are two issues right now:
1. The `advertise` call will fail if it's made before the ros connection is established.
2. The `advertise` call also expects a data definition as part of the call.

Here's some sample code that's working for me. Maybe you can follow this example as a temporary workaround while I look into a nicer fix for this??

```typescript
import { PanelExtensionContext, RenderState } from "@foxglove/studio";
import { useLayoutEffect, useEffect, useState, useCallback } from "react";
import ReactDOM from "react-dom";

function PublisherExample({ context }: { context: PanelExtensionContext }): JSX.Element {
  const [renderDone, setRenderDone] = useState<(() => void) | undefined>();

  // We use a layout effect to setup render handling for our panel. We also setup some topic subscriptions.
  useLayoutEffect(() => {
    // The render handler is run by the broader studio system during playback when your panel
    // needs to render because the fields it is watching have changed. How you handle rendering depends on your framework.
    // You can only setup one render handler - usually early on in setting up your panel.
    //
    // Without a render handler your panel will never receive updates.
    //
    // The render handler could be invoked as often as 60hz during playback if fields are changing often.
    context.onRender = (renderState: RenderState, done) => {
      // render functions receive a _done_ callback. You MUST call this callback to indicate your panel has finished rendering.
      // Your panel will not receive another render callback until _done_ is called from a prior render. If your panel is not done
      // rendering before the next render call, studio shows a notification to the user that your panel is delayed.
      //

      console.log(`Got render state: ${renderState}`);
      // Set the done callback into a state variable to trigger a re-render.
      setRenderDone(done);
    };

    // After adding a render handler, you must indicate which fields from RenderState will trigger updates.
    // If you do not watch any fields then your panel will never render since the panel context will assume you do not want any updates.

    // tell the panel context that we care about any update to the _topic_ field of RenderState
    context.watch("topics");

    // tell the panel context we want messages for the current frame for topics we've subscribed to
    // This corresponds to the _currentFrame_ field of render state.
    context.watch("currentFrame");
  }, []);

  // invoke the done callback once the render is complete
  useEffect(() => {
    renderDone?.();
  }, [renderDone]);

  const onPublishClick = useCallback(() => {
    if (context.advertise) {
      console.log(`About to advertise example topic`);
      try {
        context.advertise("/foxglove/example_publisher", "std_msgs/String", {
          datatypes: new Map(
            Object.entries({
              "std_msgs/String": { definitions: [{ name: "data", type: "string" }] },
            }),
          ),
        });
        console.log("Advertise succeeded");
      } catch (err) {
        console.log(`Caught err while advertising: ${err}`);
      }
    } else {
      console.log("Cannot advertise example publisher");
    }
  }, [context]);

  const onButtonClick = useCallback(() => {
    if (context.publish) {
      console.log(`About to publish from example publisher`);
      try {
        context.publish("/foxglove/example_publisher", { data: "Hello world! @" + new Date() });
      } catch (err) {
        console.log(`Got error while publishing: ${err}`);
      }
    } else {
      console.log(`Example publisher cannot publish`);
    }
  }, [context]);

  return (
    <>
      <div>Click button to advertise</div>
      <button onClick={onPublishClick}>Advertise</button>
      <div>Click button to publish</div>
      <button onClick={onButtonClick}>Publish Message</button>
    </>
  );
}

export function initExamplePanel(context: PanelExtensionContext) {
  ReactDOM.render(<PublisherExample context={context} />, context.panelElement);
}
```

ccpowers, 2022-01-05:
@foxymiles Thanks, I will try this out! 
The `datatypes` argument to the advertise call isn't in the extensions panel api methods doc (https://foxglove.dev/docs/extensions/panel-api-methods) - it just says there's an optional `options` argument - might be good to add some more detail there. 
Another question I have regarding the api - is there any return from the `context.advertise` call that I could use to know if the advertising of the publisher has succeeded? Or any introspection into what topics the panel currently has advertised?

foxymiles, 2022-01-05:
> @foxymiles Thanks, I will try this out! The `datatypes` argument to the advertise call isn't in the extensions panel api methods doc (https://foxglove.dev/docs/extensions/panel-api-methods) - it [...]

Yes you're right we should document that `datatypes` parameter better and perhaps also make it easier to use. 

That last patch I merged should make the handling of the `advertise` call more robust. We're discussing now ways we might make the `advertise` and `publish` APIs a bit more ergonomic for developers.

foxymiles, 2022-01-07:
@ccpowers I've merged some changes that should make it possible for you to call `advertise` directly in `useLayoutEffect` as in your original code sample instead of waiting for user input. The `datatypes` parameter would still be required as in my example. I'm going to expand on the documentation for this method in our extension API to make it more clear how this parameter should be used.

foxymiles, 2022-01-07:
Closing this for now. Please let me know if you're still having trouble @ccpowers.

closed 2022-01-07 by foxymiles