foxglove/studio #2557: Remove background thread performance warning

amacneil, 2022-01-01:
**Description**
When opening Studio in Safari, I get a toast warning "Your browser does not support rendering charts in a background thread". However this pops up directly over top of another warning "You’re using an unsupported browser". We don't need both.

- Version: latest main
- OS: web

[image]

jtbandes, 2022-01-03:
> We don't need both.

I thought our plan was to remove the persistent banner: https://github.com/foxglove/studio/issues/1422#issuecomment-1000555742
When that's gone, will you still want to omit the performance warning?

closed 2022-01-03 by jtbandes

amacneil, 2022-01-04:
@jtbandes once FF + Safari are "fully supported" (whatever we define that to mean) we should remove the persistent banner for those browser (and keep it for people using Lynx or whatever browsers are left).

> When that's gone, will you still want to omit the performance warning?

That's an interesting question, I guess it depends how bad the performance is. Figma manages to run on FF + Safari without any performance warnings. I kinda feel like at the point we officially mark FF + Safari "fully supported" we should probably do away with any performance warnings and just make our app work as best we can in those environments? But open to thoughts.