foxglove/studio #2572: When I open local xxx.bag file using foxglove, how can I build the communication between info of xxx.bag and other local program

noone3001, 2022-01-05:
Dear unselfish developer：
When I open local xxx.bag file using foxglove, I would like to use the info of xxx.bag to debug my local program. Then I find out that  foxglove can decode my xxx.bag and apparently the massages flow inside foxglove. I'd like to know how to build the communication between info of xxx.bag and other local program.
Best wishes, thank U.

jtbandes, 2022-01-05:
Hi, what do you mean by “info of xxx.bag”? What info do you want to use other than the message data?

Foxglove Studio uses our TypeScript library [`@foxglove/rosbag`](https://www.npmjs.com/package/@foxglove/rosbag) to read bag files.

noone3001, 2022-01-05:
Like we using ROS command to replay bag file `rosbag paly xxx.bag`, then I can use the recorded message to debug my program. Foxglove can only display the message, can not let message to debug my program. That is, there is no master like ROS master to manager the communication between the recorded message and my program.