foxglove/studio #2577: Playback speed control not working when loading from data studio

ccpowers, 2022-01-05:
**Description**
When playing back a bag loaded from the data platform, the playback speed controls do nothing. If I load a bag stored locally, they do what I expect. 

- Version: Mac Desktop 0.25.0, also Chrome
- OS: MacOS, also happens in Chrome on MacOS

**Steps To Reproduce**
1. Load a bag from the timeline view of data platform
2. Attempt to change the playback speed

https://user-images.githubusercontent.com/477928/148280268-bc9db58b-ffa4-49db-ab11-2876e4b76f61.mp4

**Expected Behavior**
Playback changes speed, and UI updates to reflect this.

**Actual Behavior**
Playback appears to remain at 1x (time advancing 1 second per clock second) and the dropdown option does not change, even if you click on different speeds.

jtbandes, 2022-01-05:
Thanks for the report!

closed 2022-01-05 by jtbandes

ccpowers, 2022-01-10:
Thanks for the quick follow up! Is there a specific release cadence that I can use to figure out when this will be available, or is it CI-d into the web version or anything?

amacneil, 2022-01-10:
The web version hosted at [studio.foxglove.dev](https://studio.foxglove.dev/) is always up to date with main, we deploy continuously.

Desktop releases happen every 1-2 weeks (sometimes sooner if we have high priority bugs to deploy). This PR was included in v0.26 - we keep release notes [here](https://github.com/foxglove/studio/releases) 🙂.