foxglove/studio #2594: Make "loop" default on, remember it as a global app preference

amacneil, 2022-01-06:
**Description**
- Enable "loop" by default
- Discussion topic: Should we remember "loop" state across restart of the app? Ideally it wouldn't sync across tabs, but if I close and restart the app I would get my last state rather than always resetting to the default.

amacneil, 2022-07-06:
@defunctzombie thoughts on this?

I think making it a global app setting should be non controversial. Could go either way turning it on by default.

defunctzombie, 2022-07-07:
User profile would be a decent place if we don't need the user to "toggle" it in app settings - and I don't see why we would since they can toggle it in the playback controls.

defunctzombie, 2022-07-07:
> @defunctzombie thoughts on this?

Seems reasonable. Added a good first issue tag. Just another issue in the long list.

amacneil, 2022-08-15:
Closing due to lack of user interest. If people feel strongly about remembering the loop state, please feel free to chime in.

jtbandes, 2022-08-15:
Personally I find it annoying to manually turn on the loop setting every time. But I might not be a regular user since I'm mostly loading data for testing purposes only.

amacneil, 2022-08-15:
I feel similarly, but Roman preferred having it always default to non-loop like video players do. I don't have a strong argument either way so I think this one is just waiting for more data points.

jhurliman, 2022-10-14:
Please fix this

defunctzombie, 2022-10-14:
Sounds like there's more preference for remembering this setting globally for the app so let's make that the plan. Do we want to move the `playback speed` setting to an app setting? This is a layout setting. It does seem a bit weird to have these two stored in different places.

amacneil, 2022-10-23:
I think playback speed could also be set free from the layout. I agree it seems weird if some playback controls are part of the layout and others are separate. It seems reasonable that all playback controls can be independent from the layout and persisted as app preferences.

cc @banisadr