foxglove/studio #2609: Arrow markers ignore frame_id when defined with two points

jhurliman, 2022-01-07:
**Description**

The storybook `/story/panels-threedimensionalviz--arrow-markers` demonstrates the bug. When arrows are defined with pose+scale, they render properly. When an arrow is defined with two points, the frame_id is ignored because the pose field is never read. The relevant code is in <https://github.com/foxglove/regl-worldview/blob/main/src/commands/Arrows.js#L43-L55>

[image]

defunctzombie, 2022-01-21:
@jhurliman sending your way per our conversation in slack

jhurliman, 2022-01-28:
Fixed in https://github.com/foxglove/regl-worldview/pull/13

closed 2022-01-28 by jhurliman