foxglove/studio #2613: Bring back the car model

jhurliman, 2022-01-07:
Requiring external hosting of mesh markers is burdensome. It would be nice to bring back the embedded autonomous vehicle glTF, and allow referencing it with a special URL such as `foxglove://car`.

amacneil, 2022-01-07:
> ) or Android (
> ID: <foxglove/studio/issues/2613 @ github. com>

Maybe we offer a few options in a dropdown or something?

Although, do people actually want built in models that are not their robot? I can't think of who would want this (besides us developing FG).

Sent via Superhuman ( ***@***.*** )

jhurliman, 2022-01-08:
@amacneil I was thinking this would be for mesh markers, although it could be used for `PoseStamped` as well with a dropdown in the UI.

> do people actually want built-in models that are not their robot

We already supply a built-in model that is not their robot. It's a low poly cylinder / arrow thing. This request is to expand support beyond our single model to at least one more.

amacneil, 2022-01-10:
Still don't understand who this would be useful for. Pretty big difference between supporting generic shapes and a "Chevvy Bolt" model.

jhurliman, 2022-01-13:
@amacneil deeproute.ai

[image]

You can see in their promo video they use a generic car model that does not appear to be an accurate model of a fully rigged self-driving car. They thought a generic car model made for better data visualization than a purple cylinder driving down the road. To get to a generic car model in Studio, you can:

1. Acquire a 3D model of your vehicle, often derived from a CAD model or another very high poly source
2. Hire an artist to create a low poly rendition of this model
3. Convert the model to glTF format, ensuring you set the correct up axis and scale, the file size is reasonable, and no unsupported glTF extensions are enabled
4. Upload this model to a publicly accessible web server and configure the correct CORS headers. Alternatively, if you want to keep your model private you need to run a private web server hosting this file
5. Point your mesh_resource at this public or private URL

With an embedded car model, you could get to the same visual fidelity of the deeproute.ai promo video by:

1. Point your mesh_resource at `foxglove://car`

rgov, 2022-04-10:
What about making it easy to select Ignition Fuel models? https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Hatchback%20red