foxglove/studio #2630: Missing transform error when I'm not using transforms or coordinate frames at all

jhurliman, 2022-01-09:
**Description**

When publishing `sensor_msgs/PointCloud2` message over the ws-protocol that does not have a `frame_id` field in header, I get the following error:

[image]

**Steps To Reproduce**

Publish a message over ws-protocol, or encode a message into MCAP that is recognized as a renderable topic in the 3D panel but omit `header.frame_id` from the message definition.

**Expected Behavior**

I would expect my point cloud to render, and not see any errors about missing transforms since I am not using any transforms.

defunctzombie, 2022-01-11:
Dug into this error and created a WIP PR to show the failure in a story (https://github.com/foxglove/studio/commit/d68f2aa58cfb4d2f6d0a8c680ac207d4a52985f4). The larger issue here is that we do not support omitting a frame_id.

Our type information for header assumes a frame_id field is present so we'd need to relax that to find the places in 3d panel where we make such assumptions. I imagine this work is related to empty string frames_id, tho that creates its own amusing side-effects in our app.

defunctzombie, 2022-01-11:
@jhurliman Gave it some more thought and I think the PR I've drafted might be workable given our current 3d panel structure. There might be some other things to tidy up but give it a try with your workflow and see if it behaves as expected.

commit a8f19279 (2022-01-11): Default makers with no frame_id to the empty frame

When computing marker pose, a marker with an empty frame (or no frame) exists in the (empty) frame. If we are rendering in the (empty) frame we can return the marker pose rather than performing transformations.

Fixes: #2630
commit 424657f1 (2022-01-11): Default makers with no frame_id to the empty frame (#2642)

When computing marker pose, a marker with an empty frame (or no frame) exists in the (empty) frame. If we are rendering in the (empty) frame we can return the marker pose rather than performing transformations.

Fixes: #2630
closed 2022-01-11 by defunctzombie