foxglove/studio #2645: Rosbridge: Support reading and setting ROS params

jtbandes, 2022-01-11:
Studio users connected to a data source via a Rosbridge connection (ROS 1 or 2) should be able to:
- Receive and read ROS params from their ROS stack
- Set ROS params by sending values back to the stack

Interpause, 2022-06-30:
Based on <https://github.com/RobotWebTools/roslibjs/issues/419#issuecomment-1116704060>, it might be possible now for ROS2 with rosbridge?

bertaveira, 2022-07-13:
This would be a very useful feature for when using studio to monitor live a remote device (since native ros2 connection normally has problems opposed to rosbridge). We currently ssh to the device which is not the cleanest or fastest way to change parameters

Interpause, 2022-07-14:
Modifying node parameters are actually service calls in ROS2. So https://github.com/foxglove/studio/pull/3552 should make it possible to adjust parameters for ROS2. Just waiting for an extension or official panel for service calls now...

danclapp4, 2022-09-08:
@Interpause @bertaveira Great news, I have written a custom extension to do this exact thing for ros2! with a rosbridge connection, you can view, set, and load ros2 parameters.  I am currently also trying to work on being able to save your parameters to a yml file for if you like your configuration, but want to keep messing with parameters.  

Check it out! https://github.com/danclapp4/ros2-parameter-extension