foxglove/studio #2646: Autocomplete in Node Playground `Input<>` throws an error

jtbandes, 2022-01-11:
**Description**
This is a regression compared with webviz.io/app.

**Steps To Reproduce**
- Add a new node in Node Playground
- Place cursor inside `Input<|>`, type quotes `Input<"|">`, and press ctrl+space to trigger autocomplete.

**Expected Behavior**
[image]

**Actual Behavior**
[image]

defunctzombie, 2022-01-16:
It seems that the generated roslib injection is causing monaco editor to receive a new roslib:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/NodePlayground/Editor.tsx#L84-L89

I replaced the above code with:

```typescript
React.useEffect(() => {
    if (rosLib.includes("Placeholder")) {
      return;
    }
    const disposable = monacoApi.languages.typescript.typescriptDefaults.addExtraLib(
      rosLib,
      `file:///node_modules/@types/${projectConfig.rosLib.fileName}`,
    );

    return () => {
      disposable.dispose();
    };
  }, [rosLib]);
```

And that makes it more obvious that monaco editor (language server?) is not taking into account the new library code.

defunctzombie, 2022-01-16:
The above extra lib issue seems to be related to https://github.com/microsoft/monaco-editor/issues/2866

And may or may not be the issue described in this bug report. Still investigating.

PR #2687 (2022-01-18): Downgrade monaco-editor to 0.30.1
**User-Facing Changes**
Node playground editor correctly updates when a new ros library is generated.

**Description**

This fixes a bug with loading the generated rosLib into monaco editor. The monaco editor language server does not pick up the newly generated roslib.

This change adds a check to webpack.ts to alert us on future upgrades to confirm the feature works.

See: https://github.com/microsoft/monaco-editor/issues/2866

Fixes: #2646
commit 698bbb8f (2022-01-18): Downgrade monaco-editor to 0.30.1 (#2687)

This fixes a bug with loading the generated rosLib into monaco editor. The monaco editor language server does not pick up the newly generated roslib.

This change adds a check to webpack.ts to alert us on future upgrades to confirm the feature works.

See: https://github.com/microsoft/monaco-editor/issues/2866

Fixes: #2646
closed 2022-01-18 by defunctzombie