foxglove/studio #2651: Remote file does not support HTTP Range Requests when I got a file from url

yh1215853664, 2022-01-11:
**Description**
I'm using docker On my Server.
When I point the "Open File from URL",I got an error,Remote file does not support HTTP Range Requests
But I have confired that the server support range request.
The follow is the result on my client

PS C:\Users\Administrator> curl -i http://file.shui-cn.com/1.bag -H "range:bytes=1024-2000"
HTTP/1.1 206 Partial Content
Server: nginx/1.16.1
Date: Tue, 11 Jan 2022 20:58:35 GMT
Content-Type: application/octet-stream
Content-Length: 977
Last-Modified: Tue, 11 Jan 2022 20:44:03 GMT
Connection: keep-alive
ETag: "61ddec13-7b5bae1"
Accepr-Ranges: bytes
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, OPTIONS
Access-Control-Allow-Headers: DNT,X-Mx-ReqToken,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Authorization
Content-Range: bytes 1024-2000/129350369

And The button of Open local file is not avaliable when I using docker on my server
I will get the error from the brower console 

OpenDialog.tsx:52 
        
       TypeError: window.showOpenFilePicker is not a function
    at Uo (showOpenFilePicker.tsx:13:25)
    at useOpenFile.tsx:28:32
    at OpenDialog.tsx:50:7
    at Zu (react-dom.production.min.js:219:180)
    at oc (react-dom.production.min.js:259:160)
    at S.unstable_runWithPriority (scheduler.production.min.js:18:343)
    at eo (react-dom.production.min.js:122:325)
    at io (react-dom.production.min.js:252:279)
    at Es (react-dom.production.min.js:243:371)
    at react-dom.production.min.js:123:115
Thanks a lot

- Version:.026.0
- OS:Centos7

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

yh1215853664, 2022-01-11:
I try it both on Microsoft Edge and Google Chrome.The Version is 97.0.1072.55(Edge) and 96.0.4664.93(Chrome)

jtbandes, 2022-01-11:
`Accepr-Ranges: bytes`

Is this a typo or is the server really returning an `Accepr-Ranges: bytes` header? It should be `Accept-Ranges: bytes`.

yh1215853664, 2022-01-11:
Sorry It is a typo 
The follow is my nginx conf

    server {
       listen       80;
       server_name file.shui-cn.com;

      location / {
        alias /opt/bag/;
        aio on;
        directio 512;
        sendfile on;
        autoindex on;
        autoindex_exact_size off;
        autoindex_localtime on;
        add_header Accept-Ranges bytes;
        proxy_force_ranges on;
#        add_header Cache-Control no-store;
        add_header Access-Control-Allow-Origin *;
        add_header Access-Control-Allow-Methods 'GET, POST, OPTIONS';
        add_header Access-Control-Allow-Headers 'DNT,X-Mx-ReqToken,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Authorization';
        if ($request_method = 'OPTIONS')
                {
                return 204;
                }

           index index.html index.htm;
      }
   }

As for the issue for browser,I cannot work in Chrome, the same wrong with Edge
>

yh1215853664, 2022-01-11:
Sorry It is a typo
The follow is my nginx conf

server {
   listen       80;
   server_name file.shui-cn.com;

  location / {
    alias /opt/bag/;
    aio on;
    directio 512;
    sendfile on;
    autoindex on;
    autoindex_exact_size off;
    autoindex_localtime on;
    add_header Accept-Ranges bytes;
    proxy_force_ranges on;
    add_header Cache-Control no-store;
    add_header Access-Control-Allow-Origin *;
    add_header Access-Control-Allow-Methods 'GET, POST, OPTIONS';
    add_header Access-Control-Allow-Headers 'DNT,X-Mx-ReqToken,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Authorization';
    if ($request_method = 'OPTIONS')
            {
            return 204;
            }

       index index.html index.htm;
  }
}

As for the issue for browser,I cannot work in Chrome, the same wrong with Edge

defunctzombie, 2022-01-11:
How are you running Studio? Are you using the desktop app or the web app?

> And The button of Open local file is not avaliable when I using docker on my server

Could you clarify what you mean by this? What is running in docker?

defunctzombie, 2022-01-12:
We looked into this and found the cause of the confusing error message. For your nginx config you need to expose the `Accept-Ranges` header for CORS.

```
add_header Access-Control-Expose-Headers 'Accept-Ranges';
```

We've also made an enhancement to the error message to make this more clear. https://github.com/foxglove/studio/pull/2653/files

closed 2022-01-12 by defunctzombie

yh1215853664, 2022-01-14:
> nges';

Dear,I use the docker displayed on github home page that recommend,then I proxy 8080 port by My Nginx.
I can't point the button "Open Local File" on my web browser. The error I show it above.

yh1215853664, 2022-01-14:
> We looked into this and found the cause of the confusing error message. For your nginx config you need to expose the `Accept-Ranges` header for CORS.
>
> ```
> add_header Access-Control-Expose-Headers [...]

Thanks, It works.
Then I want to ask an question,
when I got an http file through the button "Open file from url",
It show an error."Requested more data than cache size: 8692015104 > 209715200"
I want to know why ?
Limit it from nginx or the code limit it ?

defunctzombie, 2022-01-15:
This sounds like an issue with Studio itself. Here is where that line of code comes from:

https://github.com/foxglove/studio/blob/35a752505267d438d83aa2b65236cb8c4877a69e/packages/studio-base/src/util/CachedFilelike.ts#L162

I don't have an answer on why this happens tho. Are you able to share your file (privately via email is ok) and I can spin up a server and try to reproduce. It might be something with how Studio is trying to read the remote file.

PR #2653 (2022-01-12): Improve error message for failed remote bag requests

**User-Facing Changes**
When failing to load a remote url from a server, the user is shown an error message with more details on what might be misconfigured.

**Description**
In the web app, remote bag requests may fail due due to CORS errors. This change improves the error message to call out which parts of the CORS configuration should be verified to ensure compatability with studio.

Fixes: #2651
commit e43f01ac (2022-01-12): Improve error message for failed remote bag requests (#2653)

In the web app, remote bag requests may fail due due to CORS errors. This change improves the error message to call out which parts of the CORS configuration should be verified to ensure compatibility with studio.

Fixes: #2651