foxglove/studio #2669: Selecting some plot series adds an extra period to the message path

amacneil, 2022-01-13:
**Description**
Adding a plot series that ends with `.x` (and some other suffixes) causes the series to be added with an extra period, creating an invalid message path input that has to be fixed manually.

https://user-images.githubusercontent.com/637671/149415324-291c05c1-0ee7-48b7-909b-da1561cfd4a6.mov

foxymiles, 2022-01-14:
@amacneil Can you still repro this in current main? I'm not seeing it here.

amacneil, 2022-01-14:
I just did some further testing, it only happens when you use find-as-you-type. It doesn't happen if you only use the mouse and select suggestions from the dropdown.

foxymiles, 2022-01-14:
> I just did some further testing, it only happens when you use find-as-you-type. It doesn't happen if you only use the mouse and select suggestions from the dropdown.

Thanks for the extra detail that helps.

PR #2690 (2022-01-18): Use message path type to drive autocomplete.
**User-Facing Changes**
This improves message path autocomplete behavior.

**Description**
The approach here is to calculate what the new path would be when an autocompletion is selected and determine if it would result in a path to a simple value. If so we don't append a `.` to help the user continue searching.

Fixes #2669
closed 2022-01-18 by foxymiles