foxglove/studio #2671: ros2 remote bag

rogue-shadowdancer, 2022-01-14:
I've been using webviz and foxglove's remote bag feature for ros1 for a long time, it's a great feature! I just need to store the bag on the nas, and I can open it through the url and view it, awesome!
I am going to try to use ros2 recently, can I ask if Foxglove has plans to develop remote bags for ros2 bags?

amacneil, 2022-01-14:
The default ROS 2 bag format (sqlite) is not well-suited for remote access over a network, it would be very slow due to how the file is laid out on disk.

There are two solutions we are working on:

- [MCAP](https://github.com/foxglove/mcap) is a new robotics recording file format for we are developing for both ROS 2 and non-ROS robots. It is currently in draft so not ready for production, but it will provide several improvements over sqlite, for example much easier to read over a network. Please provide any feedback on this and you are welcome to test it out once libraries are available in the coming weeks.
- Our [Robotics Data Platform](https://foxglove.dev/data-platform) can store your data in the cloud, and we automatically optimize storage and indexing so that we can provide very fast access in Foxglove Studio. We don't currently support importing ROS 2 bag files, but happy to discuss if this is important to you (feel free to reach out via https://foxglove.dev/demo or https://foxglove.dev/slack and we can chat more).

rogue-shadowdancer, 2022-01-17:
Cool!
But, there is a problem that I am working for a company, which means most of our data storage method is inconvenient to change (because many people are using it), and it cannot be uploaded to the cloud (because of confidentiality).
Anyway, I will discuss related solutions with the leader, thank you for your help!

lanyusea, 2022-01-17:
@amacneil 

Hi Adrian, wondering if there are some benchmark data to see how much baudwidth it need to support a remote rosbag2.

It will be great if there are any patch/brach supporting the remote rosbag2 feature.
We'd like to do a test with our ceph first if possible.

amacneil, 2022-01-17:
Others at Foxglove might be able to add some more technical details, but my high level understanding is that there are two problems with remote ROS 2 sqlite bag files:

1. We would have to download the whole file before we can read any of the data (unlike ROS 1 bags where we can read them as a stream and show data as it appears).
2. Due to browser limitations, we would need to download this whole file into memory, so the file size would be limited by how much memory a browser tab is allowed to use (about ~2gb I believe).

As mentioned we developing MCAP which will be available for ROS 2 and solve these problems. Also, our data platform supports on-premises data storage if you do not want to use the cloud. Feel free to [reach out](https://foxglove.dev/demo) if you are interested in our data platform or paid support.

We can keep this issue open, and accept a PR if someone wants to try adding remote ROS 2 bag support, but due to the limitations above inherent to the ROS 2 sqlite file format it may not be a good user experience, and only work for small files.

lanyusea, 2022-01-18:
@amacneil totally understand, thanks for the reply

jtbandes, 2022-09-08:
The MCAP project has matured a lot and is now available as a rosbag2 storage plugin. We recommend that folks use this going forward: https://github.com/ros-tooling/rosbag2_storage_mcap

closed 2022-09-08 by jtbandes