foxglove/studio #2677: Pixhawk5 log file cannot be opened, parsing error

jhurliman, 2022-01-14:
[log_3_2022-1-13-15-10-20.ulg.zip](https://github.com/foxglove/studio/files/7872834/log_3_2022-1-13-15-10-20.ulg.zip)

```
Error initializing player: Cannot read 2 bytes from 11927552 byte source, 0 bytes remaining
```

This file can be correctly parsed and rendered at https://logs.px4.io/

jhurliman, 2022-01-14:
The attached log file appears to be truncated. The final three bytes of the file are: `0x2f 0x00 0x44`

0x2f00 - A 47-byte message follows (uint16 little-endian)
0x44 - This is a message of type `DATA`

Anything less than 47 more bytes is an invalid message, especially 0 bytes since all DATA messages begin with a two-byte `msgId` field which are the two bytes that can't be read in the error message above. I can look at improving the robustness of the log parsing to still extract partial data out of truncated log files, if that's the root cause of what happened here.

vj2181, 2022-01-17:
It is very likely that the file was truncated as power was not turned off cleanly. During in field tests , it is very likely for that to be case as well. So as you said, we are perhaps looking for improved robustness to files not closed cleanly.

vj2181, 2022-01-18:
If it is helpful at all - removing the last two bytes (as indicated in the error message) in the file and saving it made it work. Hope this is helpful for the fix.

closed 2022-01-18 by jhurliman