foxglove/studio #2697: No live plotting when connected to ROS2 via webbridge

mhl787156, 2022-01-19:
**Description**
Hi, I'd like to tune a PID controller within foxglove connected to my live ROS2 system via the ros-web-bridge. However it seems that I cannot get anything to come up in the plot. I'm fairly sure i've followed instructions but they only ever refer to plotting from a rosbag. A gif is attached here showing the value i'm trying to plot changing live, and the plot (which should at least show something):

[image]

Apologies in advance if i've missed a 'start' or 'enable instruction somewhere!

I have tried with a number of different topics, but nothing seems to come through! 

- Version: Foxglove Studio version 0.27.0, Ubuntu Desktop Version
- OS: Ubuntu 20.04

**Steps To Reproduce**

1. Start a live connection to ROS2 via the webbridge
2. Create a plot and point it to some data. Observe that nothing is happening
3. Use a raw panel to observe that data is actually coming through

**Expected Behavior**
The live plotting of data points on the plot as it scrolls along

**Actual Behavior**
Nothing gets plotted

jtbandes, 2022-01-19:
Hi, when you say ros-web-bridge are you referring to [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) or [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite)? The rosbridge suite is actively maintained and is highly recommended over the ros2-web-bridge.

mhl787156, 2022-01-19:
Hi thanks for the reply - sorry I forgot that there were two, we are using rosbridge_suite! The 3d viz and raw data streams through quite nicely, only have problems with the plotting.

jtbandes, 2022-01-19:
I'm not able to reproduce this problem on my VM – I tested using `ros2 topic pub /foo geometry_msgs/msg/PoseStamped`. Is there more info about your setup that you can share / steps to reproduce?

https://user-images.githubusercontent.com/14237/150198414-8b810b46-d843-4a6f-a5c1-4491ecf5556b.mov

mhl787156, 2022-01-19:
Interesting, thanks for testing that out. I will see if I can get a minimal example of it not working for you, as well as do a bit more investigation. It's evening where I am at the moment, so I will get back to you in the morning. 

For now, if its not too much hassle, I am essentially running this docker-compose file below which runs a bunch of drone related things (needs renaming back to txt because github wont allow yaml files..). Then I connect foxglove desktop to localhost:9090. The only oddity of note (although it hasnt caused trouble with other foxglove stuff) is that all mavros topics are run through the ros1_bridge. 

[docker-compose.windows.yaml.txt](https://github.com/foxglove/studio/files/7900027/docker-compose.windows.yaml.txt)

mhl787156, 2022-01-19:
I also tried just publishing using the cli tools and for some reason I dont appear to getting anything either. I am running everything inside docker containers (rosbridge, ros2 foxy etc.) and they must be communicating since I can see the packets in the raw output.  

https://user-images.githubusercontent.com/10742598/150200784-72364912-5b07-4b9a-a18c-bb0c35a91e90.mp4

jtbandes, 2022-01-19:
Hm, I tried `docker compose -f docker-compose.windows.yaml up` but I'm not able to connect the websocket server:

```
gh2697-rosbridge-suite-1  | [rosbridge_websocket-1] [ERROR] [1642620660.487521003] [rosbridge_websocket]: [Client 1] [id: call_service:/rosapi/topics_and_raw_types:2] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist
```
I'm guessing this is the source of the problem (the rosapi node doesn't launch correctly):
```
gh2697-rosbridge-suite-1  | [rosapi_node-2] [WARN] [1642620711.028101638] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

jtbandes, 2022-01-19:
Another thing I would recommend trying is to record a bag and try playing back the bag directly in Studio. If it has the same problem then you can just share the bag with us and that will make debugging much easier – if not, then at least that narrows things down a little.

mhl787156, 2022-01-19:
Okay looks like its something on my side for now, I will investigate further and get back to you! Thanks for your help so far :)

jtbandes, 2022-01-19:
👍  feel free to join us in Slack as well: https://foxglove.dev/slack

mhl787156, 2022-01-20:
Okay I have narrowed it down to a problem where one of my ROS2 containers (running gazebo + other bits and bobs) is causing the plotting to just stop for all values when it was working just before. 

If I then stop the container, the plotting does not come back until I refresh foxglove. 

@jtbandes  I was wondering if you might be able to point me to some foxglove logs or rosbridge_suite logs to see what this container might be interfering with? And do you have any clues as to what might be happening - would it perhaps be an unrecognised msg type (I do have ros-foxy-gazebo-msgs installed with rosbridge_suite)? 

https://user-images.githubusercontent.com/10742598/150369163-47b3f1f6-eae6-49e8-9547-32dff62fc909.mp4

Note I found today that I am running an older version of rosbridge_suite (v1.0.7), but updating to the latest on apt (v1.1.0) broke everything - but thats a different issue for now! 

And thank you for the slack invite, I was going to post there with the video, but it said you guys had run out of space! So thought I might continue here instead.

mhl787156, 2022-01-20:
Diving into the container, it is looking like gazebo and specifically the backend of gazebo ([gzserver.launch.py](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/dashing/gazebo_ros/launch/gzserver.launch.py) is causing that cutoff to happen.

mhl787156, 2022-01-20:
Apologies, final update - I have managed to remap `/clock` to `/gazebo_clock` and lo and behold, the plotting remains and time does not get reset! 

Unfortunately, I need `/clock` for all of the other modules that are in use in my system it would seem, so remapping is not really a viable solution (unless I remap all of the other clocks, or find a way to remap it specifically for foxglove)

[image]

I assume foxglove automatically takes `/clock` as gospel on time and got confused at whatever gazebo was sending it. Perhaps an addition to foxglove might be to let the user select where it gets it time from (or where it doesnt want to get its time from!).

On that note, before I published on `/clock` where was foxglove getting its time from? 

(Also renaming issue)

defunctzombie, 2022-01-24:
From some of the above videos it looks like the issue might be related to jumping back in time. What happens if you open Studio after launching gazebo? I'm trying to isolate if the issue is whenever gazebo is publishing /clock or when gazebo starts publishing /clock after Studio is open and connected; thus changing the time window for Studio.

mhl787156, 2022-01-24:
Hey, okay so I first started gazebo and then the publisher, then opened up studio. Immediately the clock shows start time as 13.8... like so:
[image]
But no plotting occurs. My guess is that the timestamp given in the Raw (16x10^10 ish) is what is used for plotting... and doesnt match the time given by Gazebo on /clock (and therefore used as foxglove default). 

A colleague of mine flagged these issues as potentially related:
foxglove/studio#1293
foxglove/studio#2322

defunctzombie, 2022-01-27:
Its possible that the `ros2` live data source doesn't have any support for `/clock`. All the `clock` logic seems to be commented out in the ros2 player:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/Ros2Player.ts#L563-L577

@jhurliman Sending your way to on how we want to handle this in the ros2 player. At the very least we should inform users if `/clock` is an unsupported feature.

defunctzombie, 2022-02-10:
@mhl787156 could you give the latest web version a try? We recently merged https://github.com/foxglove/studio/pull/2801 which is aimed at fixing some outdated `/clock` handling in the Rosbridge player connection.

mhl787156, 2022-02-10:
@defunctzombie I can confirm that the 2D line plotting is now working on 0.29.0-dev! 

[image]

Whereas it still was broken on the latest desktop release (0.29.0?). 

Unfortunately I seem to have lost 3D plotting now! I am unsure if its related or not. All I can tell is that if I bypass the use of /clock (gazebo node remap from /clock to /gazebo_clock) is that the 3D works in 0.29.0-dev and 0.29.0. But if  I do not have the remapping, nothing comes up on the 3D. Either way plotting now works though so thanks for that fix!

defunctzombie, 2022-02-10:
@mhl787156 In that last screenshot with the 3d panel issue - could you expand the top level `header` field of the raw message panel? What we are looking for is that the stamp in that header field is before the `end time` of the data source.

mhl787156, 2022-02-10:
@defunctzombie Here you go:

[image]

The fast_motion_planner/path has a ros2 timestamp as a header. Interestingly in the bottom left, we are plotting local_position/pose and showing corresponding the raw PoseStamped which has the same stamp format as the path topic. msgs.

mhl787156, 2022-02-10:
@defunctzombie Ah that makes a lot of sense. Is there a way of either the 3d panel plotting by `receive time`? Or a method of choosing the time source so I can ignore `/clock` and use ROS2 default time then?

closed 2022-02-10 by defunctzombie

mhl787156, 2022-02-10:
Okay thanks for your help!