foxglove/studio #2706: Collada assets not loading properly

reinzor, 2022-01-20:
**Description**

For the robot description visualization, we use collada assets. In Blender or RVIZ, these assets are being loaded properly:

**Blender**

[image]

**RVIZ**

[image]

- Version: 0.27.0
- OS: Ubuntu 20.04

**Steps To Reproduce**

Example collada file:
[lift.zip](https://github.com/foxglove/studio/files/7906483/lift.zip)

**Expected Behavior**

Render properly with correct orientation and materials

**Actual Behavior**

Renders like this:

[image]

jtbandes, 2022-01-22:
Hi, regarding the positioning issues, I don't think the .dae file you provided is enough to reproduce the problem. I was able to get it to display in the 3D panel, but it looks like the issues you're showing are related to other models that are also in the scene with different transforms? Any chance you can provide a sample bag that contains all the data necessary to reproduce?

[image]

reinzor, 2022-01-24:
[lowpad_description.zip](https://github.com/foxglove/studio/files/7926849/lowpad_description.zip)

Attached a description package for our m1 lowpad.

```
roslaunch lowpad_description rviz.launch
```

[image]

I added the package since it cannot be reproduced with a bag file since it is loading assets via the `ROS_PACKAGE_PATH`.

jtbandes, 2022-01-24:
I've determined that the issue with incorrect colors is blocked by https://github.com/mrdoob/three.js/issues/21538.

jtbandes, 2022-01-27:
@reinzor Update: we've learned that the orientation issue stems from the use of `<up_axis>Z_UP</up_axis>` in your collada assets. It turns out that RViz **does not** respect the `up_axis` setting and only supports the default value of `Y_UP` (https://github.com/ros-visualization/rviz/issues/1045). This has led to most URDFs in the wild being written with the assumption of a `Y_UP` up axis for collada models (because that's how RViz renders them), regardless of the actual `up_axis` in the model, which may vary depending on the modeling tool that was used to author the model.

Because we know that many URDF packages depend on this rendering quirk (we've tested several of them in Studio), we are planning to add a setting to Studio's 3D panel to enable RViz compatibility by ignoring the `up_axis` and `unit` settings in collada models. However, since in your case you're using a custom lowpad_description package rather than an off-the-shelf robot description from a third party, we would recommend that you instead **change your collada models** to use the default up_axis of `Y_UP`, and change your URDF files accordingly. This will give you the best chance at compatibility with Foxglove Studio, RViz, and any other rendering tools.

reinzor, 2022-01-28:
Thanks for the update, we'll look into it. Weird that both Blender and RVIZ are not respecting this setting then.

jtbandes, 2022-01-28:
Blender probably respects the up_axis, which is fine if you are looking at a collada mesh alone. The problem is when your URDF (which places the meshes relative to each other) is based on an assumption that the renderer will ignore the up_axis.

closed 2022-01-28 by jtbandes