foxglove/studio #2714: 3D panel doesn't use tf to transform marker

chenjunnn, 2022-01-21:
**Description**
When publishing a marker in a frame different from the fixed frame, the 3d panel doesn't transform its position to the fixed frame but showing its position directly.

As the example below, the marker should be in front of the camera_link (x=0.4), but it is incorrectly shown at z=0.4 in camera_link.

[image]
(The left is camera_link, and right is the optical frame)

jhurliman, 2022-01-21:
I see `frame_locked` is set to false. Is it possible the relevant tf was published after the marker?

We have a number of screenshot tests to confirm that transforms are applied to markers correctly in a variety of different tf hierarchies. It's certainly possible that you found a gap in our tests, but at first glance this seems like something that should have test coverage and be working correctly.

chenjunnn, 2022-01-21:
The tf is static, and here is what I see using Rviz
[image]

defunctzombie, 2022-01-23:
@chenjunnn Am I correctly seeing in your first screenshot (of Studio) That the transforms are correctly shown in the scene (the various arrows) relative to each other?

Which type of datasource are you using?

chenjunnn, 2022-01-23:
The transforms are correctly shown. And I'm using ros2 rosbridge connection. But I think the problem has nothing to do with the datasource.
I also record a example rosbag, the marker is incorrectly shown in studio either.
[rosbag2_2022_01_23-15_00_29.zip](https://github.com/foxglove/studio/files/7919970/rosbag2_2022_01_23-15_00_29.zip)

defunctzombie, 2022-01-23:
@chenjunnn Thanks for providing the sample bag file. I've been able to open it and also see what you describe with the marker shown at the wrong location.

@jhurliman The tf_static messages appear before the marker message in the data source.

foxymiles, 2022-01-25:
Looks like this is a bug in regl-worldview's handling of some marker types and  not related to the datasource type or the topic name.

PR #2723 (2022-01-25): Correctly apply orientation to cube lists and sphere lists
**User-Facing Changes**
This fixes an issue with the rendering of some marker types in the 3d view.

**Description**
This uses an updated version of `regl-woldview` that correctly transforms markers that use `points`.

This story screenshot demonstrates spheres positioned either via the `pose` or via `points` aligning correctly on the transformed axes.

[image]

Fixes #2714
closed 2022-01-25 by foxymiles