foxglove/studio #2715: Cannot get data from rosbridge_suite v1.1.0

mhl787156, 2022-01-21:
**Description**
I just upgraded my container's rosbridge_suite from v1.0.7 to latest on apt v1.1.0 and foxglove seems to have stopped working... It registers as a client, but doesn't seem to receive anything as shown here:
[image]

Previously when running v1.0.7 I would get something like the following. This also includes an extra topic I manually published over `ros2 topic pub /test_topic geometry_msgs/msg/Point {}`:
[image]

I have double checked with my own frontend running roslibjs connecting to the same rosbridge suite seems to work fine for both the older and newer versions. 

For context, I have rosbridge_suite running as a Docker Container running this simple container: https://github.com/StarlingUAS/ProjectStarling/blob/master/system/rosbridge-suite/Dockerfile 

It is possible that I might have misunderstood something or provided the wrong configuration. But I have found that the configuration is reproducable on other machines (with the docker file below at least). I have not attempted to test outside of the dockerfile natively as I do not have the correct machines! 

Many Thanks in advance

- Version: 0.27.0
- OS: Windows & Linux & Chrome

**Steps To Reproduce**

Testing rosbridge_suite v1.1.0:
1. Run the following container: ` docker run -it --rm --network test_default -p 9090:9090 uobflightlabstarling/rosbridge-suite:latest`. This runs the dockerfile linked above. 
2. In a seperate terminal, setup a rostopic pub by running ` docker run -it --rm --network test_default  uobflightlabstarling/rosbridge-suite:latest bash`, Source ros (`. /opt/ros/foxy/setup.bash`) and run `ros2 topic pub /test_topic geometry_msgs/msg/Point {}`
3. Open Foxglove desktop or web and connect to rosbridge instance on `localhost:9090`. 
4. You should see that rosbridge_suite has a new client connected and the foxglove does not complain, but it is all waiting for data.

To test the working rosbridge_suite v1.0.7, you can use the same containers except with the tag `v0.1.0` instead of `latest`. 

Inside the containers you can check the version of rosbridge-suite by using `dpkg -s ros-foxy-rosbridge-suite | grep Version`

**Expected Behavior**
Foxglove connects succesfully and shows the topics in the second image, as well as displays the `/test_topic` topic

**Actual Behavior**
Foxglove connects successfully but it is show to be waiting for data.

defunctzombie, 2022-01-23:
After looking into this I believe the issue is with the [getTopicsAndRawTypes ](https://github.com/foxglove/roslibjs/blob/develop/src/core/Ros.js#L676-L679) call. Studio calls this method on our roslib client when first establishing a connection. This method is suppose to invoke the rosapi service call to get the list of available topics and datatypes.

Using your example Dockerfiles, I've run the following command in your ros2 environment:

```
root@a25d31d0def1:/# ros2 service type  /rosapi/topics_and_raw_types
rosapi_msgs/srv/TopicsAndRawTypes
```

Which returned a response of `rosapi_msgs/srv/TopicsAndRawTypes`. This does not match the [serviceType](https://github.com/foxglove/roslibjs/blob/develop/src/core/Ros.js#L679) hard-coded in roslibjs.

If I try to invoke the service directly using `ros2 service call` and use the roslibjs hard-coded service type then I get the following error:

```
root@a25d31d0def1:/# ros2 service call /rosapi/topics_and_raw_types rosapi/TopicsAndRawTypes
Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.10', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/command/service.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/verb/call.py", line 58, in main
    return requester(
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/verb/call.py", line 69, in requester
    module = importlib.import_module('.'.join(parts[:-1]))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'rosapi.srv'
```

You mentioned in your report that you are able to successfully use roslibjs. Would it be correct to say that your use of roslibjs does not make the `getTopicsAndRawTypes` method call? If you can try making this method call and seeing if it stalls for your use of roslibjs that would be interesting to know.

Regarding this issue; I can confirm that Studio is not able to complete the proper initialization routine it expects when connecting to the latest version of rosbridge suite on ros2 foxy due to the stalling `getTopicsAndRawTypes` call.

The issues to fix and improve are:
- Fix roslibjs to send the appropriate `serviceType`. This might be tricky since ros1 and ros2 behave differently but this is something we need to solve.
- Fix rosbridge_suite to report an error when the service call fails
- Optionally, add a timeout to Studio for the `getTopicsAndRawTypes` call so we can detect in Studio when the roslibjs call stalls. This is defensive since roslibjs and even rosbridge_suite should be updated to detect the failed call.

defunctzombie, 2022-01-23:
/cc @jtbandes who's worked on rosbridge_suite to double check my findings and add any additional thoughts.

mhl787156, 2022-01-23:
> You mentioned in your report that you are able to successfully use roslibjs. Would it be correct to say that your use of roslibjs does not make the getTopicsAndRawTypes method call? If you can try [...]

Yes thats right, in our own (very simple) [version](https://github.com/StarlingUAS/starling_ui_example/blob/main/ui/src/ros_buttons.js) we only call the very basic elements (connect, publish, subscribe etc.). Thanks for looking into it, odd that nobody else seems to have flagged it yet!

jtbandes, 2022-01-24:
The type name mismatch is happening because the types were moved to a separate package in https://github.com/RobotWebTools/rosbridge_suite/pull/665.

However, I think this is a red herring. As far as I can tell, when the rosbridge receives a service call from the client, it ignores the `type` sent by the client: https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/rosbridge_library/src/rosbridge_library/capabilities/call_service.py#L59-L68

And when I run `ros2 launch rosbridge_server rosbridge_websocket_launch.xml` (running the server from source) Studio is able to connect.

I can reproduce the issue with Studio being unable to connect when using your Dockerfile. I think the root cause is https://github.com/RobotWebTools/rosbridge_suite/issues/704.

mhl787156, 2022-01-24:
I might have gotten versions mixed up, but the dockerfile running version 1.0.7 of rosbridge (tag: `v0.1.0`) does connect and send data successfully to Foxglove despite doing the double /rosapi thing. 

The dockerfile running 1.1.0 of rosbridge (tag: `latest`) doesnt connect but also has the double /rosapi thing. 

----

So I found a linux laptop and tried building from source. The latest master works absolutely fine as you said. However! rosbridge_suite version 1.1.0 specifically does not work. I'm guessing that whatever happened there was a fix between then and the newest version. 

The reason im using 1.1.0 is because it seems to be the only one available on apt-get ubuntu 20.04 (as of a few days ago). Perhaps the easiest solution is to update to the latest (either on apt or myself building from scratch) and that might solve all the problems!

defunctzombie, 2022-01-24:
[This commit](https://github.com/RobotWebTools/rosbridge_suite/commit/c624f97f0b7ea8ee64386790b0ed99d81d01fbee) came after 1.1.0 was released and looks related.

jtbandes, 2022-01-24:
> [Just checked there are only 1.1.0 -1focal on apt for ros-foxy-rosbridge-suite and the other rosbridge elemetns]

The apt repo is updated during semi-regular "syncs". It looks like the last one for Foxy was Dec 2, so I imagine there will be one soon, though it doesn't seem to be announced yet. You can follow https://discourse.ros.org/c/release/foxy/65 and see the build/release status at http://repo.ros2.org/status_page/ros_foxy_default.html?q=rosbridge

mhl787156, 2022-01-26:
Okay thanks, for now I will build rosbridge from scratch, but it looks like it will all be resolved with the next foxy sync. 

Thanks for the help! Happy for this issue to be closed :)

commit 758e365c (2022-01-31): Add timeout to RosbridgePlayer getTopicsAndRawTypes

Some versions of Rosbridge Suite would hang on getTopicsAndRawTypes requests. This change adds timeout logic and problems reporting so the user
has feedback on what went wrong. Previously this hang would provide no feedback and the user would be left with no indication on why studio is not displaying their data.

Fixes: #2715
PR #2746 (2022-02-01): Add timeout to RosbridgePlayer getTopicsAndRawTypes

**User-Facing Changes**
There is additional user feedback if a Rosbridge connection is unable to lookup topics and raw types.

**Description**

Some versions of Rosbridge Suite would hang on getTopicsAndRawTypes requests. This change adds timeout logic and problems reporting so the user
has feedback on what went wrong. Previously this hang would provide no feedback and the user would be left with no indication on why studio is not displaying their data.

Fixes: #2715
commit f8efd7ed (2022-02-01): Add timeout to RosbridgePlayer getTopicsAndRawTypes (#2746)

Some versions of Rosbridge Suite would hang on getTopicsAndRawTypes requests. This change adds timeout logic and problems reporting so the user
has feedback on what went wrong. Previously this hang would provide no feedback and the user would be left with no indication on why studio is not displaying their data.

Fixes: #2715
closed 2022-02-01 by defunctzombie