foxglove/studio #2718: 3D scene Markers disappear randomly

sionyx, 2022-01-24:
**Description**

- Version: 0.27.0
- OS: MacOS Desktop or Web

**Steps To Reproduce**
- Foxglove ws protocol
- Send messages with MarkerArray containing 2 markers (mesh and text)
- send rate: 1 message per 0.1 second 
- video play rate is 0.01x, but is behaves the same on play rate 1.0x 

https://user-images.githubusercontent.com/1351998/150753513-9e281566-eb5c-4733-af38-7afe31d2486b.mov

Timestamp of markers same with timestamp of message

[image]

Timestamp of message is set in message frame header:

[image]

**Expected Behavior**
- Markers are always visible on scene

**Actual Behavior**
- Markers are disappear and appear again in random moments.

**Comments**
- it is not related to loading mesh model - it loads successfully ones (checked in Chrome Developer Console) and text marker disappears at the same time.
- it is not related to tf - as Roman said: I also see you are rendering in the Innopolis frame which matches your marker frame_id so I would agree tf is not related to the issue here.
- it is not related to play rate - same behaviour on different rates

Discusion archive: https://foxglove.slack.com/archives/C028UEY858S/p1642942399069300

defunctzombie, 2022-01-24:
@sionyx In your video above there is a problem indicator for the data source:

[image]

What are the problems the data source is reporting?

sionyx, 2022-01-24:
> What are the problems the data source is reporting?

@defunctzombie sometimes my backend advertise tf and tf_static channels twice with the same channel_id. Seems that it should not lead to discussed problem.

defunctzombie, 2022-01-24:
> @defunctzombie sometimes my backend advertise tf and tf_static channels twice with the same channel_id. Seems that it should not lead to discussed problem.

Can you fix that up and see if the problem still happens? Changing _tf_ is something that will impact the overall rendering of the 3d panel.

I've also made an example websocket server: https://gist.github.com/defunctzombie/83110a6314e4ccb8d278a5acc77effe1 where I've attempted to reproduce the publish pattern I am seeing in your video. This server publishes a marker message every second with a stamp incrementing by 0.01 seconds. Using this server I've not been able to reproduce your issue. This makes me think its not just how the markers are published but how other messages are impacting the 3d panel.

defunctzombie, 2022-01-24:
For additional reference here is the layout I am using:

[image]

sionyx, 2022-01-24:
Let me some time to play with your test server. I'll respond as soon as posible

sionyx, 2022-01-25:
@defunctzombie Roman, I finally modified your server to let it play my messages. Actually just 2 messages is enough to reproduce flashing. You can find them in file [ws-dump-datalines-hex](https://gist.github.com/sionyx/1d2476ca2cd88a16a119e6dd17a5c89e#file-ws-dump-datalines-hex). First line leads to hide marker, second one leads to show it again.

Here is gist with updated server and dump-file https://gist.github.com/sionyx/1d2476ca2cd88a16a119e6dd17a5c89e

I compare two messages and the only difference are positions and orientations. Positions are pretty same and look reasonable. May be something wrong with orientations?

defunctzombie, 2022-01-25:
Thanks for providing that additional info and example.

As we discussed in Slack, the marker disappears because its `header.stamp` time is _after_ the `end time` on your data source. The 3d panel considers this marker outside the time bounds and removes it.

Closing this since we've found the root cause. Feel free to re-open if you find that markers disappear even when their `header.stamp` time occurs within the dataset time bounds.

closed 2022-01-25 by defunctzombie

jtbandes, 2022-01-25:
> BTW there is some error in console:
> [image]
> It would be nice to show WebGL error somewhere.

If this error is reproducible could you please file a new issue with steps to reproduce?

sionyx, 2022-01-25:
@jtbandes I've seen this message couple of times in console by chance. If I will found way to reproduce I file issue. And actually I haven't seen errors itself, just this message about too many errors.