foxglove/studio #2722: Sample layout should overwrite existing sample layout

amacneil, 2022-01-24:
**Description**
My sample layouts are getting out of control:

[image]

Currently we create a new layout (with the same name) every time you click "view sample data". We originally did this because we were worried about overwriting the layout for a user who may have modified it. However, I think the more common scenario is someone loading the sample data multiple times, and we should just overwrite their existing sample layout with the latest version.

amacneil, 2022-05-04:
[image]

jtbandes, 2022-05-06:
Another possible solution is to switch to an existing layout with the same name, but not overwrite, only create a new one if it doesn’t exist already.

defunctzombie, 2022-05-07:
Here is where the logic lives that handles layout _things_ attached to a data source factory:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/components/PlayerManager.tsx#L161-L177

amacneil, 2022-05-08:
Let's try @jtbandes's suggestion. If the user wants to restore the original sample layout, they can delete the sample layout and it will get recreated. Otherwise, we just switch to their existing sample layout.

amacneil, 2022-05-08:
@foxymiles want to take this one?

foxymiles, 2022-05-09:
@amacneil Sure, I'll have a go.

PR #3339 (2022-05-13): Fix sample layout duplication issue
**User-Facing Changes**
This fixes an issue with proliferation of sample layouts when repeatedly loading the sample data set

**Description**
The solution here is to look for an existing layout with the same name as our sample layout and load that instead of creating a new layout. This implementation requires loading the entire list of local layouts in order to look for one with a matching name but maybe in practice this won't be a performance issue since it seems unlikely that someone would have a very large number of layouts and also often be loading the s [... cut]
closed 2022-05-13 by foxymiles