foxglove/studio #2726: Data Source: open file from url(http/https) does't support searchParams

Elaine87, 2022-01-25:
**Data Source: open file from url(http/https) does't support searchParams**

- Version: 0.27.0
- OS: macos Big Sur

**Steps To Reproduce**

[image]

**Expected Behavior**
can be opened regardless of whether url has searchParams or not

**Actual Behavior**
it has an error

----------------------------------------------------------------------------------------
I browse the code and I think the question may be in this:
 file location: foxglove/packages/studio-base/src/components/OpenDialog/Remote.tsx

```js
 const extension = path.extname(currentUrl);
```
if the variable currentUrl has searchParams which contains dot(.) will lead to above error.
Whether the pathname should be extracted as the path.extname() params:
```js
  const url = new URL(currentUrl);
  const extension = path.extname(url?.pathname);
```

foxymiles, 2022-01-25:
Thanks for the bug report and the suggested fix!

closed 2022-01-25 by foxymiles

Elaine87, 2022-01-26:
Thank you for your efficient support！Thanks for this awesome project!

PR #2730 (2022-01-25): Fix open url parsing logic to handle query params etc.
**User-Facing Changes**
This fixes an issue in the the open remote location dialog.

**Description**
This uses the `URL` class to parse the provided URL string instead of trying to extract the extension directly.

Fixes #2726