foxglove/studio #2742: 3D panel Grid - Line size scales with the grid size

jhurliman, 2022-01-27:
**Description**

[image]

**Steps To Reproduce**

[image]

foxymiles, 2022-01-28:
Looks like maybe the issue is that regl-worldview interprets `scale.x` as a thickness value?

https://github.com/foxglove/regl-worldview/blob/7364e5142b50e7d50194321af91d6bb1428f493f/src/commands/Lines.js#L298-L300

jtbandes, 2022-01-28:
I believe the grid uses `scaleInvariant` lines and this is a bug with the implementation of scale-invariant rendering.

foxymiles, 2022-01-28:
Actually according to @jtbandes this is a longstanding issue in regl-worldview and not easy to fix.

defunctzombie, 2022-01-31:
@jhurliman What is the intended behavior for `Line width`?

jtbandes, 2022-01-31:
Line width is intended to be specified in pixels.

defunctzombie, 2022-01-31:
> Line width is intended to be specified in pixels.

Why not units in the real world? meters?

foxymiles, 2022-01-31:
> > Line width is intended to be specified in pixels.
> 
> Why not units in the real world? meters?

For a reference grid you would want line width in screen units not real units right?

PR #2770 (2022-02-02): Render grid using native GL lines, remove line width setting
**User-Facing Changes**
Improved 3D panel grid rendering for large grids; removed the line width setting for grids.

**Description**
Closes #2742.

In investigating this ticket, we learned that the problem with our scale-invariant line rendering is that the [two](https://github.com/foxglove/regl-worldview/blob/92df7aad8d28d15a811d5daaed672261c507e27a/src/commands/Lines.js#L173-L176) [places](https://github.com/foxglove/regl-worldview/blob/92df7aad8d28d15a811d5daaed672261c507e27a/src/commands/Lines.js#L144-L145) where we manually mult [... cut]
commit 120542fd (2022-02-02): Render grid using native GL lines, remove line width setting (#2770)

**User-Facing Changes**
Improved 3D panel grid rendering for large grids; removed the line width setting for grids.

**Description**
Closes #2742.

In investigating this ticket, we learned that the problem with our scale-invariant line rendering is that the [two](https://github.com/foxglove/regl-worldview/blob/92df7aad8d28d15a811d5daaed672261c507e27a/src/commands/Lines.js#L173-L176) [places](https://github.com/foxglove/regl-worldview/blob/92df7aad8d28d15a811d5daaed672261c507e27a/src/commands/Lines.js#L144-L145) where we manu [... cut]
closed 2022-02-02 by jtbandes