foxglove/studio #2743: 3D panel Grid - Grid is not centered

jhurliman, 2022-01-27:
**Description**

This is most obvious with small grids (~1m or less).

[image]

**Steps To Reproduce**

[image]

defunctzombie, 2022-01-27:
Not centered in what sense?

jtbandes, 2022-01-31:
Can't reproduce. I think you must have panned your camera or have been following some frame that was not centered at (0,0,0) in world space prior to enabling the grid.

https://user-images.githubusercontent.com/14237/151877775-f73f483b-6ed8-4922-b5fd-9aec4e508306.mov

jtbandes, 2022-01-31:
Also tried a 0.1m width:

https://user-images.githubusercontent.com/14237/151878267-f8e356da-9fa9-4cb0-b544-106c33bac7eb.mov

closed 2022-01-31 by jtbandes