foxglove/studio #2745: Treat protobuf strings as uint8 arrays in PointCloud2

sionyx, 2022-01-27:
**Description**
Sending point cloud to FG via ws protocol using protobuf cause FG crash.

- Version: 0.28.0 Desktop
- OS: MacOS

**Steps To Reproduce**
Originally PointClouds2 has data in uint8array 
[image]

In [ros-message-schemas](https://github.com/foxglove/ros-message-schemas/blob/main/proto/ros/sensor_msgs/PointCloud2.proto) data field is defined as bytes which is std::string in c++. So FG receives following message:
[image]

String is not encoded with base64. It is originally bytes array.
Here is hex dump of websocket message:
`01 01000000 802b68278db8a416 0a10085d120a0d09ba40611540cf23181a001001180222090a017810001807200122090a017910041807200122090a017a10081807200122110a09696e74656e73697479100c1807200128003010382042202f5d9342fed44441ec511e41000038426d679242f4fd4441c1ca2f41000030424801`

And this leads to crash the whole app:
[image]

**Expected Behavior**
Process strings byte array where it is supposed to be byte array.

**Actual Behavior**
App crashes.

defunctzombie, 2022-01-27:
I see that according to the [protobuf language guide](https://developers.google.com/protocol-buffers/docs/proto3) that bytes does get represented as a `string` type in c++. Probably because in c++ a `std::string` is semantically equivalent to `vector<char>` (https://en.cppreference.com/w/cpp/string/basic_string).

It looks like above - something has turned the string (which should be a container of bytes) into a base64 string before sending on the wire. Is this a protobuf serialization option? Or something else? The data in this field should not be turned into a base64 value.

sionyx, 2022-01-27:
@defunctzombie as I wrote above it is not encoded as base64, you can check hex dump from Chrome Developers Tools it contains bytes. Seems that FG shows binary byte strings as encoded as base64 in Raw Messages Panel.

defunctzombie, 2022-01-27:
@sionyx Gotcha - I missed that. I think this issue is resolved by https://github.com/foxglove/studio/pull/2735 which merged yesterday. Its not in a desktop release but is released to https://studio.foxglove.dev

sionyx, 2022-01-27:
@defunctzombie yeah, nice, it works in web version! Looking forward for next desktop release. 
(Next time before filing new issue I will check not only open issues but closed as well)

closed 2022-01-27 by sionyx