foxglove/studio #2747: WASD/Arrows Support for Teleop Panel

mcdoerr, 2022-01-27:
Our current project uses the WASD keys to drive the robot instead of clicking buttons with the mouse. We are looking to switch to Foxglove because it fits our use case well, but don't want to lose this important functionality. 

We would like to request support for driving the robot with Foxglove's teleop panel using either WASD keys or the arrow keys. The keys could map to the existing inputs for the four buttons in the screenshot when the screen is in focus. 

This would greatly reduce the difficulty of driving our robot, since it is much easier to click keys than to move the mouse, especially on faster robots.

mcdoerr, 2022-01-27:
[image]
For example, the WASD keys could map like this. The arrow keys would work just as well.

jtbandes, 2022-01-27:
Thanks for the feature request. One issue we have often run into with panel keyboard shortcuts is the issue of supporting multiple panels. If a user configures a layout with 2 teleop panels (possibly for multiple robots) which one should the keyboard shortcuts apply to? Do you have any thoughts on this from the perspective of your project?

mcdoerr, 2022-01-27:
That issue makes sense, thanks for bringing it up!

In our current simulation tool, we deal with multiple teleop panels by having the user select the panel which they are driving in with a mouse click. When they click off of the "driving panel" the keyboard inputs do not send messages until the panel is reselected/focused. If a user had multiple teleop panels, they could switch by clicking on the different panels. This also does not affect the logic for driving with the mouse clicks, which is nice.

Perhaps it would be best to have this feature turned off by default, and an option to turn it on in settings? What are your thoughts? 

Thanks for the quick response! :)

robi772, 2022-01-30:
Please ADD a STOP button into the middle.

jtbandes, 2022-01-31:
Hi @robi772, can you clarify what you would want the stop button to do? Currently, the Teleop panel only publishes messages while the buttons are pressed, so just releasing the button stops the flow of messages. Would the stop button be sending a different message?

robi772, 2022-01-31:
I like to control a robot with the turtlebot3_teleop_key, when continously publishing the linear and angular messages. You just set the speed with the buttons.