foxglove/studio #2756: please add support for loading .obj files in urdfs

jbcpollak, 2022-01-28:
currently our UR10 .urdf can't be loaded because it uses .obj meshes.

jtbandes, 2022-02-04:
Is there an example urdf and .obj file you could provide? I don't see many urdfs out in the wild with .obj meshes. Does RViz support the .obj format?

jbcpollak, 2022-02-09:
Here is the urdf with obj we are using. I found an alternative ur10 udrf with stl files that has the `<xacro:arg>` bug, but I can work around that. At this time I don't think this is a priority for us. We don't use RViz (or even have it installed really) so I can't speak to its level of support.

[UR10_description_with_obj.tar.gz](https://github.com/foxglove/studio/files/8035923/UR10_description_with_obj.tar.gz)

jtbandes, 2022-02-10:
Thanks. Are there other visualization tools which support this that you/we can cross-reference this?

jhurliman, 2022-07-14:
OBJ support is implemented for mesh markers and URDFs in New3D

closed 2022-07-14 by jhurliman