foxglove/studio #2782: Make it easier to automatically populate one or more layouts into web version of Studio, auto-connect to data source

jhurliman, 2022-02-03:
Use case: I have a robot that is sometimes Internet-denied, and I want to use Foxglove Studio as the UI into this robot. I have one or more layouts that I have created specifically tailored to this robot and exported as JSON. I want to host the web version of Studio running on my robot, and when anyone visits <https://myrobot.local/> Studio loads with one of my layouts and automatically connects to my robot's data source (for example rosbridge at <wss://myrobot.local:9090>). Optionally, other layouts are available to switch to as well.

Under the hood this could be implemented with a redirect from <https://myrobot.local/> to <https://myrobot.local/foxglove/?layout_url=...&data_source=...> or something to that effect.

RFRIEDM-Trimble, 2022-02-03:
This could serve as a nice workaround in our use case to #1511 that prevents our users from uploading layouts. If it's already on the robot (through other means), no need to upload anything.

amacneil, 2022-03-30:
Making a note that the best workaround for this today is to build your own local copy of studio, and edit the default layout:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/providers/CurrentLayoutProvider/defaultLayout.ts

I agree that it would be nice to support this directly rather than requiring you to recompile the app.