foxglove/studio #2787: Single-panel embedding with URL parameter

jhurliman, 2022-02-04:
Use case: I would like to embed Studio in a blog post on <https://foxglove.dev/blog> to interactively visualize something I am talking about, such as markers in a 3D panel. To accomplish this, I am thinking about an iframe pointed to a URL of the form:

https://studio.foxglove.dev/?layout_url=...&data_source=...&panel_id=...

Where `panel_id` uniquely references one of the panels that appears in the layout loaded from `layout_url` (a remote JSON file). I would expect this to strip away most of the UI in Studio: the sidebar, and widgets for adding/removing panels or modifying the current panel.

jtbandes, 2022-02-04:
What about just making a "frameless" mode, so the user would create a layout that has a single panel and enable frameless mode to achieve the same thing? Extracting and rendering one panel from a layout seems a bit complicated.

jhurliman, 2022-02-04:
@jtbandes you're right, I was overthinking it. `frameless=true` plus a single panel layout would be fine.

amacneil, 2022-02-04:
maybe avoid a double negative? e.g. `frame=false` or `chrome=false`... latter is more accurate but maybe confusing with the web browser of the same name.

jtbandes, 2022-02-04:
I think frameless is fine, it's kind of a common term, whereas "framed" or "frameful" is not.

defunctzombie, 2022-02-04:
Something to consider - we might want to do this on a different domain than studio.foxglove.dev. Typically it is bad form from a security perspective to allow your site with a user's credentials to be embedded in another site. In fact we plan to dis-allow embedding studio.foxglove.dev for this reason.