foxglove/studio #2791: OccupancyGrid flickering in 3D view

samiamlabs, 2022-02-04:
Hi folks!

I'm getting a kind of flickering behavior when trying to look at an OccupancyGrid in the 3D view.

[image]

- Version: 0.29.0
- OS: Ubuntu 20.04

**Steps To Reproduce**
I'm using rosbridge on ROS 2 galactic and publishing to /map every other second.

defunctzombie, 2022-02-04:
@samiamlabs Can you check that the header.stamp on your occupancy grid message comes before the `end` time shown in the data source sidebar? We've seen this flickering when your messages are stamped after the `end` time.

samiamlabs, 2022-02-04:
Yes, I'll check!

samiamlabs, 2022-02-04:
Turns out I was not setting header.stamp in my custom map publihser so it was stuck on 0.

If I set it using `msg.header.stamp = node.get_clock().now().to_msg()` it wont show up at all however.
Thinking this could be related to simulation time, but not sure. 

The timestamp in my map message is now very much larger than the `end` time. 
[image]

I tried recording it to a ros2 bag, and playing that up I foxglove worked fine without any flicker.

jtbandes, 2022-02-04:
Yes, if `header.stamp` is far in the "future" (compared to current time) then the message will not be displayed. It does seem related to simulation time, because the current/end time is ~300. Studio recognizes the `/clock` topic and will use that time as the current time if it is present. Is it possible that your `node.get_clock().now()` is _not_ using simulation time?

samiamlabs, 2022-02-04:
Yes, there is probably some trick to get the node clock to use simulation time that I'm missing.
We can close this ticket while I figure out how to do that. Seems like the issue is on my end. Thanks!

samiamlabs, 2022-02-04:
Hmm, I managed to fix the timestamps, but I'm still getting the flickering.
[image]

jtbandes, 2022-02-04:
Just to check, if you use a Raw Message panel to look at the same OccupancyGrid message, does it also flicker?

defunctzombie, 2022-02-04:
I think this could be related to https://github.com/foxglove/studio/issues/2697 - we don't yet have `/clock` support for ROS2. I think we need to be better about the messaging here within the app around this working or not.

defunctzombie, 2022-02-10:
@samiamlabs could you open the Data source sidebar so we can see the `end time` and also expand the `header` field. I'd like to see what the time of the marker is relative to the last time of the datasource.

samiamlabs, 2022-02-11:
[image]

There is no flickering anymore in the latest version! (main built from source)
I tried the web and desktop app and they both work now.

Great!

closed 2022-02-11 by defunctzombie