foxglove/studio #2792: URL param to allow loading layout from a remote URL

jtbandes, 2022-02-04:
Related: https://github.com/foxglove/studio/issues/2787

I'd like to embed Studio in my website using an `<iframe>`. I need to be able to specify a layout, but I can't control the localStorage or the user's account, so I'd like to do something like `&layout.data=<base64...>` or `&layout.url=https://example.org/mylayout.json`.

defunctzombie, 2022-02-04:
Related - loading a set of "team layouts" or "layout templates" from a url? Maybe a separate issue?

jhurliman, 2022-02-04:
Also related to #2782

jtbandes, 2022-02-04:
Hm, maybe a duplicate, although that other ticket seems to be trying to cover multiple things in one.

jameskuszmaul-brt, 2022-04-28:
+1 to this. Not sure what the reference to "team layouts" is, but being able to load in with multiple useful layouts available to the user would be great as well. But just getting a single default layout via remote url + deep linking would be amazing.

jasonimercer, 2022-05-09:
A nice blend between the old webviz feature of 1 hosted layout and the current Foxglove abilities would be to allow us to load a set of named layouts from a URL/json file.

In our webviz use-case we can launch the viewer using different layouts for realtime technical observations of the state of the robots at "dark sites" where there is no internet access. We'd like to transition to Foxglove and the ability to have curated layouts hosted on the robot and switchable at runtime would be very appealing.

amacneil, 2022-05-09:
Importing comment from https://github.com/foxglove/studio/issues/2117#issue-1048474869

> In my firm we build ROS1 and ROS2 based robots. We plan to use FoxGlove to propose some maintenance pages for our robots. Most of the time, when users will use these maintenance pages, they will be offline (without any internet connection, and we can't assume that they already had one). I would like to serve foxglove web on a webserver hosted on the robot. And I would like to have some predefined layouts hosted on the robot. These predefined layouts will be designed by the robots development teams. They haven’t vocation to be modifiable.
>
> Today I do this with webviz by using the URL parameter :"layout-url", and having some JSON layout hosted on my robot. That works like a charm. I read the shared gdoc document “FoxGlove studio deeplink”, and my use case doesn't seem to be addressed. (https://docs.google.com/document/d/1jULICvQaZbJM8Et4ylFCE7GIFwedq7YkB4NhvnpjW0k/edit#heading=h.ic2oeiqyrxid)
>
> What I plan to do is to provide a backend service (that we will publish on github (or that can be part of the foxglove repo, I don’t care) ) that will serve some layouts. This service will run on our robots. Then, we could create a new class in FoxGlove Studio that implements the IRemoteLayoutStorage Service to connect to this backend. Of course our backend service will be very simple. We don't plan to address any authentication mechanism, and it will not be designed to be hosted elsewhere than in a local area network. So, someone that needs to use a layout stored on the web and shared with teammates will continue to use your existing service.

jeffmaldo27, 2022-05-15:
+1 on the ability to load layout from a file like `&layout.url=https://example.org/mylayout.json`

camchambers, 2022-05-17:
+1 for loading layouts from a file.

maxpolzin, 2022-07-05:
Thx for pointing me  to this issue, @jtbandes. +1 for this feature. 

Is there another way to load a layout when offline?
Is there another way to share layouts with non-organization people?

adrienbarral, 2022-07-06:
@maxpolzin => Did you see that you can export Layout as a JSON file to import them later.

That fullfill a lot of offline usecase.
You can do this by right clicking on your layout, that will open this contextual menu from wich you can export :

[image]

markkuskoodi, 2022-07-15:
+1 for this feature. We would use the cloud service, but our bag files are only available in our domain which means we need to use self-hosted version of Foxglove. This feature `&layout.url=https://example.org/mylayout.json` would help us a lot in terms of sharing and providing layouts.

jhurliman, 2022-07-15:
I think this is what was implemented in this commit? https://github.com/foxglove/studio/commit/59abc5e9e6ce74866f9c827e100bfca8e39e7793

That's from a fork of studio. We would need to open a PR against foxglove/studio introducing the feature.

siddancha, 2022-07-15:
Any insight on how highly the Foxglove team has prioritized this ticket?

IMO loading layouts from a remote URL is the most useful feature that's still missing from Foxglove Studio! I have literally created multiple custom builds of Studio each with a different default layout to get around this limitation 😢. Would +1 more than once if I could!

defunctzombie, 2022-07-16:
> We would use the cloud service, but our bag files are only available in our domain which means we need to use self-hosted version of Foxglove

@markkuskoodi How does the domain where your bag files are available limit using the hosted version of studio? Also happy to take this to DM on our slack if there's more context outside this ticket.

defunctzombie, 2022-07-16:
> I have literally created multiple custom builds of Studio each with a different default layout to get around this limitation 😢. Would +1 more than once if I could!

@siddancha Where has the layout storage and sync feature of the foxglove account not met your needs that you had to make custom builds with the layout as part of the build?

siddancha, 2022-07-16:
@defunctzombie I have two use cases that are not being met:

1. Sharing visualizations publicly with users who are outside my org, or maybe don't even have a Foxglove account. They currently cannot access layouts stored in my account.
2. I want to use different layouts for different types of visualizations. It would be nice if that can happen with a single URL click, instead of the user (if they are in my org that is) needing to find and click on the right layout that I would need to communicate separately.

DM-ing you separately to give an example of my use case.